prusa3d/PrusaSlicer-settings #2: Add a README how to import the settings into Slic3r

valo, 2016-11-10:
It is not very clear how to import these settings into Slic3r? Should all the ini files be merged into one before that? If you give me some guidelines I can experiment and try to write some instructions/scripts for that.

WereCatf, 2017-02-13:
I just grabbed the latest Prusa Slic3r from https://github.com/prusa3d/Slic3r/releases and placed all the files in this repo, under Slic3r-settings/Slic3r settings MK2/, in AppData/Roaming/Slic3r and POOF -- MK2 appeared as a printer in Slic3r and the filaments were available in the settings.

bubnikv, 2017-02-13:
Yeah, we shall document that, in particular what are the paths on various operating systems.

bubnikv, 2017-02-13:
I have updated the readme.

closed 2017-02-13 by bubnikv