prusa3d/PrusaSlicer-settings #6: Speed for travel moves for MK2

yschroeder, 2017-03-27:
Sorry in advance if this is the wrong place to discuss print settings!

I am looking through your different print settings for the MK2 (which result in pretty nice prints!) and I am wondering how you came up with the chosen travel move speeds.

You have chosen:
* 180 mm/s for `0.05mm DETAIL`
* 120 mm/s for `0.10mm DETAIL`
* 120 mm/s for `0.15mm OPTIMAL`
* 120 mm/s for `0.20mm NORMAL`
* 100 mm/s for `0.35mm FAST`

To some extend I understand that for different qualities different travel move speeds make sense, reducing overall vibrations. However, I think the speed difference should be the other way around in such a case.

Can you explain the idea behind these values? I am trying to optimize my settings for print speed without loosing too much print quality and I don't want to miss the point here.

bubnikv, 2017-03-27:
Thanks for pointing that out.

The maximum is likely not reached due to the limits in acceleration, but we shall look into that, if it could increase print speed and also limit some oozing.

yschroeder, 2017-04-03:
BTW: I have no problems running at 200 mm/s for travel moves.

fiveangle, 2017-05-23:
I saw these too and couldn't fathom any explaination for the specific choices.  I've changed them all to 150mm/s and have observed no apparent changes in print quality on the profiles I use (FAST, NORMAL, OPTIMAL, and 0.10 DETAIL)

bubnikv, 2017-05-23:
The travel speed is limited by the firmware settings (maximum speed and
acceleration per axis).

To see their effect on the acceleration ramps, you may enter these values
into the calculator at the end of the following page:
http://www.prusaprinters.org/calculator/

In my opinion, with out firmware configuration it does not make much
difference, whether you set the travel speed to 80 or 150, both are
acceleration limited.

closed 2017-08-24 by jindrichbenes