prusa3d/PrusaSlicer-settings #10: Changed out Start and End G-code

oleost, 2017-07-03:
There is a couple of issues with current Start and End code.

Start:

1. Mesh leveling is done with different temperatures, given that the Pinda probe is temperatur sensative this can lead to having to re adjust Z live tune value when chaning material.

2. It is oozing all over the buil platform while doing mesh leveling because of to hot nozzle.

3. It heats the bed and the nozzle up to the finale temperatur at the same time, with this the nozzle is hot without any flow through it for uneccesary lenght. ( Example printing Colorfabb HT at 105c bed takes around 8 minutes to heat bed. )

The code I have been using succesfully fixes all these.

Start G-Code

```
M115 U3.0.11 ; tell printer latest fw version

M83  ; extruder relative mode

M104 S160 ; set extruder temp to 160C 

M140 S55; set bed temp to 55C

G28 W ; home all without mesh bed level

M109 S160 ; wait for extruder temp

M190 S55; wait for bed temp

G4 P15000 ; Wait for 15 secs

G80 ; mesh bed leveling

G1 Y-3.0 F1000.0 ; go outside print area

M140 S[first_layer_bed_temperature] ; set bed temp

M190 S[first_layer_bed_temperature] ; wait for bed temp

M104 S[first_layer_temperature] ; set extruder temp

M109 S[first_layer_temperature] ; wait for extruder temp

G1 X60.0 E9.0  F1000.0 ; intro line

G1 X100.0 E12.5  F1000.0 ; intro line

```

Also the end code I have adjust a with a small retraction to prevent oozing after print is done.

End G-Code

```
G4 ; wait

M104 S0 ; turn off temperature

M140 S0 ; turn off heatbed

M107 ; turn off fan

G1 X0 Y200; home X axis

G1 E-1 F900

M84 ; disable motors
```

https://youtu.be/L-gY3CjfJc0

bubnikv, 2017-07-05:
The problem with your solution is, it requires a filament specific "hot
enough to be soft, but cool enough to not leak" temperature. There is a
risk, that if the temperature is too low, a hard blob will prevent the
induction sensor to trigger.

oleost, 2017-07-05:
The nozzle gets cleaned before hitting print if there is anything on it. ( Which is seldom with the retract end code. )

And if the nozzle is clean up on start printing I don't know if any filament actually ozes on 160c ?

oleost, 2017-07-24:
Anyone got a chance to test it?
Or is this Github repository dead and Prusa is using other ways to develop theyr software and this is just out here because of licensing ?

bubnikv, 2017-07-24:
> Anyone got a chance to test it?

We are pursuing a different path, we have a temperature compensation of the probe in the current firmware.

> Or is this Github repository dead and Prusa is using other ways to develop theyr software and this is just out here because of licensing ?

We develop our software in the open.

oleost, 2017-07-24:
Yeah, I read about the temperature compensation, maybe it isn't working 100% since some of us still have some issues with changing z heights.

Also that was only part of the issue, the other and most important part was to prevent filament drips on the build platform while doing calibration. ( And this is a common problem )

bubnikv, 2017-07-24:
@PavelSindler Would you please comment on the proposal? How does it compare to your firmware solution? Thanks.

WereCatf, 2017-07-24:
@bubnikv I have tested the temperature-compensation and I have to say that it's not very well thought out. With e.g. PETG, there's filament oozing out of the nozzle, leaving a blob on the heatbed, then once the firmware decides it has been heating the pinda-probe long enough it it goes through its usual routine and starts moving the head around, hits the blob and often drags it along onto the bed, causing instant fail of the print from the get-go.

jeffreydevreede, 2017-09-20:
I've had exactly the same problem as @oleost (I've only recently assembled my mk2s kit, and still had this problem with the grey filament that came with the kit), so I did some testing on my own and basically changing the temperature of the nozzle to 140 degrees celsius fixes it for me, I'm sure 160 should also work. I've tested it using PLA and ABS. And since PLA is afaik the filament with one of the lowest melting points, this should fix it for most if not all filament types.

oleost, 2017-10-03:
@jeffreydevreede I dont move it inside the extruder ?

Or do you mean in the End code ?

Also I did take an `G4 P15000 ; Wait for 15 secs` so the pinda probe would stabilize, unsure if it has any gains or not but I atleast have not had any single issue with my live z height going anyware after I changed out that g-code.

oleost, 2017-12-26:
I can't believe this is not yet implement, looking at streams people still have to remove plastic droplets from printbed and before it starts printing it has a long string of plastic that's needs to be removed

This is a no brainer from a user perspective?

https://youtu.be/Dhlv8snyIV0

hughht5, 2018-01-07:
This is really great - improved my printing no end!!

JohnMurga, 2018-05-09:
I do realize that this is material specific due to the temps ...
But I have applied the ideas here to my Cura profile and it's made a huge difference ...
Thanks to all on the thread !!

probonopd, 2018-07-29:
Also see #19

tsprhdm, 2019-07-02:
> I can't believe this is not yet implement, looking at streams people still have to remove plastic droplets from printbed and before it starts printing it has a long string of plastic that's needs to be removed

Summer 2019, my studens fight the exact same issue. I come up re-inventing the wheel in another issue that has been raised in prusaslicer. 

A bit disappointing that this STILL has not been implemented for reasons unbeknownst to the rest of the world.

As i explained in :  https://github.com/prusa3d/PrusaSlicer/issues/913 , the only proper solution needs to be implemented by prusa in the official firmware. / config bundle.

I can see the point for heating the BED to actual temperature, since the bed warping must be very temperature - dependend due to the design (9 bevels all pulling to their center....)

But i cannot see a point in assuming a blob at the nozzle.  You shall clean the bed AND the nozzle BEFORE starting the job. So, if we don't melt the filament, there will be no blob.

Skaronator, 2020-11-21:
As a new Prusa i3 MK3S customer after switching from the Prusa Mini I'm a bit disappointed to see that nothing happend with this issue since 2017.

The Prusa Mini actually handles this issue (as described in the initial post here) by setting the temperature to 170°C while doing the mesh bed leveling as you can see here:

`M104 S170 ; set extruder temp for bed leveling`

https://github.com/prusa3d/PrusaSlicer-settings/blob/b121b8429525251573903721105c7e096beb7417/live/PrusaResearch/1.1.9.ini#L5229

Can we have the same feature for the i3 MK3? @rtyr

LordMelkor, 2020-12-04:
Using 75% of initial nozzle print temperature is probably a decent assumption for what is hot enough without causing a blob

rtyr, 2023-04-04:
We decided we will not change the behavior on existing models (MK2/MK3). Users who prefer alternative start g-code can create separate printer profile.

closed 2023-04-04 by rtyr