prusa3d/PrusaSlicer-settings #13: Extreme retraction after print

Thijxx, 2017-08-03:
With the default settings downloaded here for the MK2 on 1.75 PLA there is some kind of after-print issue with retraction. The wheel taking the filament eats into the filament after each print so I have to open the extruder to cut the filament and load it again. It does not seem to be influenced by more or less tension on the screws. So, printing goes well but after the printing the filament gets deformed..

m-Th, 2017-08-21:
I experience this also; however I am not so sure about the cause....

Thijxx, 2017-08-21:
For me it was solved by upgrading the firmware, resetting to factory defaults, double checking the right Slic3r settings ( I had the multi extruder by mistake on my single extruder printer) and I also reset the tension screws on the feeder, had them too tight.

jindrichbenes, 2017-08-21:
MultiMaterial settings do have different retraction and end gcode, where filament retracts about 10 cm back into 4y splitter. This might have caused the issue.

closed 2017-08-21 by jindrichbenes