prusa3d/PrusaSlicer-settings #17: Min and Max layer height defaults

Bmeist, 2017-09-13:
I noticed in the default Printer Settings --> Extruder 1 for the "Original Prusa i3 MK2" preset, the Layer height limits are set to Min: 0.07 and Max: 0.25. Shouldn't this be set to Min: 0.05 and Max: 0.35 (guessing the max based on the included 0.35 FAST print settings) ?

This kind of has me questioning all the other default settings for things such as speed. I noticed a lot of the settings there are different than the Slic3r defaults it shows when you hover over a field. For example, Perimeter speed is set to 40mm/s, but Slic3r says the default is 60.  

Thanks for the help!

bubnikv, 2017-09-14:
> I noticed in the default Printer Settings --> Extruder 1 for the
"Original Prusa i3 MK2" preset, the Layer height limits are set to Min:
0.07 and Max: 0.25. Shouldn't this be set to Min: 0.05 and Max: 0.35
(guessing the max based on the included 0.35 FAST print settings) ?

Not really. The layer height limits were optimized for the particular
extrusion width. Unfortunately the print & printer settings are not quite
independent. The 0.35 FAST draft print settings is a kind of a hack. It
prints with high layer width and an unusually wide extrusion lines. There
is no other way that the 0.4mm nozzle would print a 0.35mm layer height.

> This kind of has me questioning all the other default settings for things
such as speed. I noticed a lot of the settings there are different than the
Slic3r defaults it shows when you hover over a field. For example,
Perimeter speed is set to 40mm/s, but Slic3r says the default is 60.

The "default" shown by slic3er is a historical value chosen by the original
author of Slic3r and they are hard coded into Slic3r. We plan to adjust
these "default" settings to the defaults of the MK2 printer one day.

Bmeist, 2017-09-14:
Thank you very much for the reply and info.  So what you are saying is the true(non-hack) layer height max is 0.25?  I was hoping to figure out a 0.25 or 0.30 setting since I have some prints where the majority of the item is not visible.  Can I use the 0.20 preset and just change the layer height to 0.25?  Or are there other settings I would need to modify for it to work right?  

I still don't quite understand why the 0.07 min is not 0.05, but as long as that setting is not preventing me from doing a correct 0.05 layer height, then I guess it's ok.  I was just worried that the setting was incorrectly limiting my min layer height to 0.07.  Related to this, why is the variable layer height(awesome feature!) limited to 0.07 and not 0.05?  

Thanks again!

Joostvwillegen, 2017-09-16:
Please take a look at the layer heights for the 0.6 mm nozzle, those aren't correct if I look correct. It is now 0.07 till 0.25 but I expect it to be anything near 0.1 till 0.4 more or less?

jindrichbenes, 2017-10-09:
Bmeist
0.07 was chosen as a safe height, lower the layer height ,the more likely may complications arise, due to different filaments, partialy clogged nozzle (which may still work fine with 0.1 layer, but not with 0.05) and such.

Joostvwillegen
Thanks for pointing this out, these values will be fixed.

closed 2017-10-09 by jindrichbenes

ayberkozgur, 2018-08-10:
@jindrichbenes I'm also finding 0.07mm min layer height a bit weird as there is a built-in ultradetail setting that uses 0.05mm layer heights. What is the exact purpose of these limits? To be used in variable layer heights?

bubnikv, 2018-08-10:
The limits are used to limit the minimum layer height for support
generation and for the variable layer height.

slate0, 2021-12-11:
I'm confused about this as well.  It appears that Prusaslicer has built in presets for 0.05 and 0.30, but it also has built-in default limits of 0.07 and 0.25.  

I'm assuming that when I use the 0.30 setting, the slicer impose the 0.25 limit.  (This is because in other situations, limits will override anything else I tell the printer to do.  For example, when I want to play with my max volumetric flow, I set the max speed and acceleration to 9999 and use my flow limit to govern the final speeds.) 

In this situation, it looks like the user is given default presets that won't actually work.  I think most users will just expect 0.03 to work since it's built into the software.  Right now, it looks like the system is "setting them up" to think they're printing at 0.30 but by default the 0.30 setting is "broken" and isn't what they're actually getting.

There are no warnings or other feedback to tell the user they're not actually getting 0.30.

rtyr, 2021-12-11:
The limits applies only for supports generation (supports are using variable layer height independent from objects layer height) and for variable layer height feature. The reason for 0.07-0.25 is mainly about variable layer height feature, this range will work reliably without changing other parameters.

> There are no warnings or other feedback to tell the user they're not actually getting 0.30.

No warning needed, 0.30mm layer height is used as visible in the preview.