prusa3d/PrusaSlicer-settings #18: Add custom G-code M900 Kxx to filaments

3d-gussner, 2017-09-19:
Since Prusa firmware 3.0.12-7 the 'Linear Advance Integration' is activated.
Can you please add the custom g-code 'M900 Kxx' to the filaments?
I also think you should check with your firmware team how a printer with older firmware handles the 'unkown' M900 Kxx g-code to prevent issues with customers/prints.

ManuGithubSteam, 2017-10-27:
I secon this :-) Great Idea!

jindrichbenes, 2017-11-01:
Custom M900 Gcode added to MK2 filament settings, M900 added also to MM printer settings for now.

closed 2017-11-01 by jindrichbenes

3d-gussner, 2017-11-01:
@jindrichbenes Why did you put the Linear Advanced M900 K100 value to the printer settings? As the value differs by the used filament it should be used/added there and per know filament!
I don't know in which order the start g-code is written to the g-code file, but if there is a M900 Kxx start g-code defined in the filament settings it should overwrite any other M900 Kyy g-code coming from the printer settings!

jindrichbenes, 2017-11-01:
M900 K200 was added to **MultiMaterial** 0.4 nozle since the value needs to be quite large and as of now, this value seems to perform well on different materials. We will do specific values for different filaments on MMU if we find out it's necessary.
M900 K100 was added to **MultiMaterial** 0.6 nozzle, larger nozzles seem to need smaller linear advance.

MK2(S) got different K values for different filaments.

MK2(S) and MMU filament settings are separate.

3d-gussner, 2017-11-02:
@jindrichbenes Thanks.
Do you know if a filament settings custom g-code overwrites the printer setting g-code in Slic3r?

jindrichbenes, 2017-11-02:
Yes, but for whole print.
i.e. if you have M900 K100 in printer gcode and M900 K50 in filament gcode, start gcode will have lines

M900 K100
M900 K50

K50 being active for whole print