prusa3d/PrusaSlicer-settings #22: Why no settings for 0.2mm with 0.25mm nozzle??

boelle, 2017-10-23:
if you use the rule that you can have a max layer height of 80% of the nozzle diameter then 0.2mm layer height is just the max and should be included

what is the reason for not having it?

bubnikv, 2017-10-23:
> if you use the rule that you can have a max layer height of 80% of the
nozzle diameter

This is really a rule of thumb. With 80% the layer adhesion will suffer. I
personally would not go much above 60%.

boelle, 2017-10-23:
then it makes sense

i just copied over the 0.15 settings for 0.25 nozzle and changed to 0.2 height and it seems to work fine

only one problem  2-3mm above the bed, rest of print looks perfect

if you have setting for 0.2 and 0.25 nozzle somewhere not on github i would like to try them out if my copy of 0.15 settings is not good enough

should speed settings be different?

jindrichbenes, 2017-11-01:
You may try slightly slower speeds on 0.2 mm layer, but again, going over 0.15 layer height might give you poor results.

boelle, 2017-11-01:
i have since gone to 0.15mm 

but i still have issues with the print on layer 4-8 or close to that

if i print a marvin the legs will come apart, if i print a benchy it will seperate at the 4 pillars near the stearing house

https://photos.app.goo.gl/1rHri2n44Gqq7lXT2

closed 2019-01-16 by boelle