prusa3d/PrusaSlicer-settings #24: I have now tried printing a file 2 times sliced by slicer1.37.2 --> ends up in HALTED print

henrikkoch, 2017-11-08:
Both times the printer stops with heat ON in the print (im using Octoprint also). According to the command line in Octoprint it have an error in line 7509 in the included gcode where it says: Recv: Error:checksum mismatch, Last Line: 7509

I here include the STL file, the log from Octoprint and the GCODE file that I did via SlicerPE and my slicer config bundle

[files to report.zip](https://github.com/prusa3d/Slic3r-settings/files/1455623/files.to.report.zip)

Hope this helps to find the reason for the error

bubnikv, 2017-11-09:
I don't think this issue relates the slic3r settings, but likely either
OctoPrint or prusa-firmware.
Please post it to prusa-firmware.

henrikkoch, 2017-11-13:
I updated my printer with the new released firmware 3.1.0 and the problem seems to have disappeared. The GCODE file can now be printed without halting. :-)

closed 2017-11-13 by henrikkoch