prusa3d/PrusaSlicer-settings #31: Prusa PETG 2018 is different

joysfera, 2018-03-26:
Apparently the PETG sold by Prusa (manufactured by Plasty Mladec) is different in spring 2018 than it used to be so the existing Slic3rPE _Prusa PET_ profile does not produce good results. A blob on printer nozzle is created almost instantly during solid infill and regular infill does not even stick to the previous layer - it prefers to stick to nozzle instead. It looks a bit like over-extruding but perhaps it's just too melted?

Please note that Plasty Mladec confirmed me that the PETG is indeed different now - they had to lower the temperature while manufacturing it by 10 degrees C so perhaps the Slic3rPE settings need to reflect that?

bobek, 2018-03-26:
Just for the record - the original info, I've received from Plasty Mladec (in Czech):

>Popravdě řečeno došlo ke dvoum změnám:
>
>V půli roku 2017 se o půl procenta zvedlo dávkování barvícího koncentrátu do polymeru – zákaznický požadavek na vyšší probarvení. Toto ale nebylo u bílé barvy
>
>A od začátku roku 2018 pozorujeme mírnou změnu v dodaném polymeru od světového výrobce. Změnu pozorujeme při výrobě struny, materiál je tekutější. Museli jsme jít se zpracovatelskými teplotami o 5 - 10°C dolů. Konzultoval jsem toto z techniky od výrobce polymeru, ti o žádné změně nemluvím a nepřipouští.
>
>Zkuste tedy nižší teploty na HE, zmenšit flow…

joysfera, 2018-04-20:
Please do not modify Slic3rPE Prusa PET profile (as I suggested originally). It works flawlessly for all PETG filaments but the black spool I bought from Prusa in March 2018. I tried several spools of **Prusa PET made in 2017** and even a spool of PETG by Devil Design - they all print perfectly with the current Prusa PET profile. Since just a single spool made in 2018 is unusable it's obviously a (hopefully temporary) filament issue that I'm going to RMA.
This issue can be closed, I believe (unless there are tons of other Prusa PET complaining customers).

jindrichbenes, 2018-11-14:
Closed by request.

closed 2018-11-14 by jindrichbenes