prusa3d/PrusaSlicer-settings #32: mk3 settings seem to conservative compared to prusa control

pyalot, 2018-04-19:
I was printing the fine-feature test with prusa slic3r edition using the mk3 profile. I got a lot of stringies/webs.

The setting is on 0.8mm retraction at 35mm/s speed and rectract before wipe set to 0%: Slic3r-settings/Slic3r settings MK2S MK2MM and MK3/printer/Original Prusa i3 MK3.ini

I got much better results from prusa control and wondered why. It turns out prusa control uses [1.6mm](https://github.com/prusa3d/PrusaControl/blob/869989b9828a6cea1a5c877425ef0487b770e303/data/materials_i3_mk3.json#L1501) retraction, I suspect a higher retraction speed (probably the 120mm/s firmware max) and [60% rectract before wipe](https://github.com/prusa3d/PrusaControl/blob/869989b9828a6cea1a5c877425ef0487b770e303/data/materials_i3_mk3.json#L2458). 

Using these parameters and 240mm/s retraction speed (probably just bumps into the firmware max) I do get better results on the fine feature test. Still a few stringies but barely any webs. Not as good as prusa control, but better.

Maybe the Slic3r MK3 profile should be adjusted?

m00dawg, 2018-07-24:
Found this after some searching around as I'm having some issues with strings and whisps, mostly while trying to dial in MatterHackers BuildSeries PLA with the Mk3 (the filament included with the MK3 doesn't seem to have as much of an issue here). They're managed by way of a heat gun but my old printer didn't have this problem (it had other problems, among them was not stringing).

Any Prusa folks around to explain the differences in settings?

bubnikv, 2018-07-25:
@jindrichbenes answered, that the configuration structure of PrusaControl allows for overriding the retract settings per filament, which is currently not possible with Slic3r. So there is still some work to be done for Slic3r.

closed 2019-04-21 by rtyr