prusa3d/PrusaSlicer-settings #36: Material mass densities

triplepoint, 2018-06-16:
None of the supplied filaments have mass densities set, so none of the usage estimates end up providing a final mass estimate for the sliced part.  It's convenient to have these, so you can estimate in general terms what fraction of a spool is going to be needed for a given print.

For at least the simpler materials, it should be ok to provide generic values from typical examples.  For instance: https://www.filoalfa3d.com/en/content/16-our-materials

ABS -  1.05 g/cm^3
PLA - 1.27 g/cm^3
PETG - 1.25 g/cm^3
HIPS - 1.04 g/cm^3

While these values are probably not perfectly accurate for specific brands of filament, they're probably within a few hundreds of a gram / cm^3.

ron1kelly, 2018-07-08:
> This is very important with the new version with locked profiles. There's
> no way for me to unlock the profiles now, so I need to be able to have
> densities so can tell how much filament is going to be used. I'm going back
> to the old Slic3r PE for now.

It is working now, I shut down the printer, Slic3r and the computer last
night.  When I turned on the computer and printer and sliced a new project,
the estimated time was there.  I have not had a chance to go back into
Github and cancel the issue.

gyscos, 2018-08-02:
Different websites give different densities:
https://filament2print.com/gb/blog/16_densities-lengths-3D-printing-filaments.html

Many sources seem to give 1.24-1.25 for PLA, and 1.27 for PETG.

triplepoint, 2018-08-02:
If we just rounded to the nearest tenth of a gram/cm^3 that'd probably cover most variations between brands.  That's provide a decent starting point.

bubnikv, 2018-08-02:
We should have the densities set in the latest alpha configuration, which will come live with the upcoming Slic3r 1.41.0 release
https://github.com/prusa3d/Slic3r-settings/blob/master/live/PrusaResearch/0.2.0-alpha3.ini

Please let us know whether you have any objections / proposals.

BTW, the prices are given in US dollars, given by the prices from the Prusa3d e-shop.

triplepoint, 2018-08-31:
This looks resolved, I'll close this.  Thanks for the work.

closed 2018-08-31 by triplepoint

darBis, 2018-09-06:
Just have updated to the latest settings version in slic3r 1.41.0 and material densities for PET and PLA seem to be swapped.

Filament | Density [g/cm³]
---------|-----------
Prusa PLA | 1.24
Prusa PET | 1,27
Generic PLA | 1.27
Generic PET | 1.24

Compare values for Generic and Prusa profiles.

Also I would second that density of 1.27 for PLA in bug description is not correct, based on many datasheets of materials i have. Value of 1.24 for PLA seems to be more "generic". Same goes for PET with generic density of 1.27 g/cm³.

Please set default densities of Generic PLA and PET to same values as in Prusa branded profiles.