prusa3d/PrusaSlicer-settings #42: Question: Output option estimated time and fill-pattern

3d-gussner, 2018-08-03:
I would like let the Slic3r PE generate the filename but some parameters aren't working.

I try to generate the filename but adding {fill-pattern} or {external-fill-pattern} but get errors using these.
This works:
`{input_filename_base}_{filament_type[0]}_{layer_height}mm_{perimeters}perm_{fill_density}percent_{printer_model}.gcode`
This not:
`{input_filename_base}_{filament_type[0]}_{layer_height}mm_{perimeters}perm_{fill_density}percent_{fill_pattern}_{printer_model}.gcode`

Is it possible to use a parameter to get the estimated time in the output options filename?
At this moment i generate the gcode and than rename it manually adding the shown estimated print time.

bubnikv, 2018-08-03:
> This not:
{input_filename_base}_{filament_type[0]}_{layer_height}mm_{perimeters}perm_{fill_density}percent_{fill_pattern}_{printer_model}.gcode

What slic3r version? If you add the keys one by one, you will find out which one fails. The error message should point an arrow to the key it does not understand, but unfortunately the message is shown with variable width font, so the arrow points to a wrong place.

> Is it possible to use a parameter to get the estimated time in the output options filename?

Not yet, but it is in the working.

3d-gussner, 2018-08-04:
Slic3r Prusa Edition 1.41.0-alpha3+win64

Message is:
Slic3r has encountered an error

Failed processing of the output_filename_format template.
Parsing error at line 1: Unkown scalar variable_type
{input_filename_base}_{filament_type[0]}_{layer_height}mm_{perimeters}perm_{fill_density}percent_{fill_pattern}_...
where the ^ is in the middle of mm.

But that ^ position isn't that generates the issue, it is adding '{fill_pattern}_' or '{external_fill_pattern}_'

darBis, 2018-09-06:
I was struggling with same issue yesterday. All my attempts to add fill_pattern failed. Is it supposed to work?

Where are those names defined anyway? I was grepping trough Slic3r sources and I have not foud a definitive answer, even whether it should be "fill_pattern" or "inflill_pattern". Both are defined and used for the same thing trough out the codebase.
Maybe this is an issue with slic3r and not in settings.

My Slic3r PE version: 1.41.0+win64

manticorp, 2019-05-10:
> Not yet, but it is in the working.

Has there been any movement on this? I'd love to be able to tell how long a print is going to take from the printer before starting printing.

Gunnl, 2019-05-23:
Just update your documentation with a list of variables ...

bubnikv, 2019-05-24:
there is a wiki page at the PrusaSlicer on the macro language, which is quite up to date. also the variable names are shown at the tooltips of the ui preset editor.

you are welcome to extend the wiki

jmsaltzman, 2020-03-04:
I wonder if anybody figured out a way to get fill_pattern or infill_pattern.
Any clues would be appreciated :)