prusa3d/PrusaSlicer-settings #43: Request: Predifine elefantfoot compensation values for standard-presets

RacingHell, 2018-08-04:
That slic3r-feature is genius, since it makes postproocessing of sharp edges unnecessary. Please add those values to the standard prusa profiles. For .4 nozzle as example I figured out a moderate value of .25. 
In advance it might be superior, if the elefantfoot compensation fades out over a well defined and analyzed number of layers to get perfect geometric fidelity. Let‘s name it mammothtree-compensation to understand what I‘m meaning. :)

bubnikv, 2018-08-05:
We used to have this feature enabled by default, but the current
implementation is not working well for thin features, it reduces them
excessively. Before enabling this feature again in our default profiles, we
will need to make this feature adaptive.

rtyr, 2020-04-05:
Implemented in our default profiles. Closing.

closed 2020-04-05 by rtyr