prusa3d/PrusaSlicer-settings #44: Missing G90 in MK3 start gcode

FormerLurker, 2018-08-07:
I noticed that G90 was removed (possibly never included) from the default MK3 start gcode, though there are G1 commands that expect absolute positioning in there.  This seems dangerous to me, since the printer axis mode could be relative before the print begins.  I do see an M83, but I feel a G90 needs to be in there too.  I didn't check the other profiles either, only the MK3, so this might be the case for other printer profiles too.

Can you explain why you left the G90 out in the first place?  I'm interested because I'm writing plugins for OctoPrint (Octolapse), and use the G90 to detect the XYZ (sometimes E) axis mode.  If there were a way to query the current axis mode(s) from the printer I wouldn't have to worry about this, btw :)

Thank you!

rtyr, 2019-05-07:
Hi. We will add G90 to the start g-code. It is already included in latest settings bundle for Slic3rPE 1.42.0+ (beta2 at the moment).

closed 2019-05-07 by rtyr

FormerLurker, 2019-05-07:
Thank you so much @rtyr!  You just made my day.