prusa3d/PrusaSlicer-settings #45: End Code and Start Code Improvements

GreatGrizzly, 2018-08-14:
This issue was raised for the Slic3r, but this also applies to Slic3r-settings:

Frankly, the current start gcode and end gcode suck. The start gcode causes plastic to get all over the bed, attach to the nozzle, and is just inefficient time wise for starting prints.

The end gcode is also pretty bad, leaving no room to clean nozzles, or letting plastics bend up and stick to the nozzle when they cool (Very prevalent with PETG and TPU).

Here is the following GCode I use that minimizes mess, eases cleaning, and streamlines the the printer start state and end states. I think something similar to the following should be part of the default settings. Input is welcome to try to get this as generic as possible for most printer setups:

Start GCode

M115 U3.3.1 ; tell printer latest fw version
M83 ; extruder relative mode
M140 S[first_layer_bed_temperature] ; set bed temp, continue on
M104 S{first_layer_temperature[0]-70} ; set extruder temp to 70° less than the target temp to prevent oozing, continue on.
G28 W ; home all without mesh bed level
G1 Z150 X125 Y-2 F{machine_max_feedrate_y[0]*60} ; get head above bed to keep PINDA probe cool (needed for MK2), move nozzle up and away from left side camera to make cleaning easier. I don't want to raise all the way because this might cause new filament spools to come undone which can tangle.
M190 S[first_layer_bed_temperature] ; wait for bed temp.
G28 Z ; home z without mesh level again, as bed temp has changed
G80 ; mesh bed leveling
M104 S[first_layer_temperature] ; set extruder temp to first layer temp, continue on
G1 Y-3.0 F{machine_max_feedrate_y[0]*60} ; go outside print area
M109 S[first_layer_temperature]; wait for extruder temp
G92 E0
G1 X60.0 E9.0 F{machine_max_feedrate_e[0]*9}; intro line at max mm/s
G1 X100.0 E12.5 F{machine_max_feedrate_e[0]*9}; intro line at max mm/s
G92 E0

End GCode

G4 ; wait for buffer to clear
G92 E0; set extruder value back to 0
G1 E-1 F100; retract 1mm at speed of 100mm/s to help prevent a string at the end of print.
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X125 Y210 F4800; Bring print surface forward as fast as possible per min, center x to try to keep filament on spool and hotend cable off usb plug
G1 E10 F900; Crank out some plastic to weigh down the plastic thread. This prevents material from curling up as it cools, sticking to hotend.
M104 S0 ; turn off hotend
G1 Z205 F1200; Raise to top. Because this is centered in previous gcode, the filament spool should be directly above, so the risk of unraveling the spool should be lower.
M84 ; disable motors

bubnikv, 2018-09-01:
@jindrichbenes Would you please test the conditional G-code to lower the nozzle temp before mesh bed leveling? This is a long standing request and it is worth evaluating. Thank you.

jindrichbenes, 2018-09-03:
First of all, we are hardly ever able to replicate the oozing issue. Secondly, lowering temperature of the nozzle and heating up again adds a couple of minutes to the start print time, which is generally unpleasant for users, to some, mehs bed leveling itself takes too long as it is. Third issue would be that should some plastic be on the nozzle anyway, there might be an issue with this plastic compromising mesh bed leveling. This does not happen with hot nozzle. 

Still, we will look again into this and try to come up with an improvement.

rennne, 2019-02-07:
> First of all, we are hardly ever able to replicate the oozing issue.

Try with moist filament and/or use a higher printing temperature.

> Secondly, lowering temperature of the nozzle and heating up again adds a couple of minutes to the start print time, which is generally unpleasant for users, to some, mehs bed leveling itself takes too long as it is.

Waiting untill 5 degree less before starting mesh bed leveling could give the time back needed for that extra heating time.

M190 S[first_layer_bed_temperature-5] ; wait for bed temp minus 5 degree, still continues heating during following mesh bed leveling.
G28 Z ; home z without mesh level again, as bed temp has changed
G80 ; mesh bed leveling
M104 S[first_layer_temperature] ; set extruder temp to first layer temp, continue on
M190 S[first_layer_bed_temperature] ; wait for bed temp.

> Third issue would be that should some plastic be on the nozzle anyway, there might be an issue with this plastic compromising mesh bed leveling. This does not happen with hot nozzle.

I would consider that as the biggest problem. Maybe M104 S{first_layer_temperature[0]-70} ; is not the best approach when we consider that a filament usually can be printed in a  temperature range like (190 C-230 C). So minus 70 would be 120C - 160C which could result in hard or soft depending on what the user had chosen. 

I have the impression that when filaments have the tendency to ooze it usually starts at the same temperature somewhere about 190C independent of the material (PLA, PETG, PC, ..). 
So maybe setting it to 180C could be a solution. I guess most filaments should be soft enough to not cause trouble, but also not ooze at this temperature. (there are filament that print colder than 180C)

M104 S{min(first_layer_temperature[0], 180} ; set extruder temp to max of 180 °C prevent oozing, continue on.

marcosscriven, 2021-05-24:
Please could I bump this one? I just took a look at the code for the MK3, and note it hasn't changed in four years? 

https://github.com/prusa3d/PrusaSlicer-settings/blob/master/old/Slic3r%20settings%20MK2S%20MK2MM%20and%20MK3/printer/Original%20Prusa%20i3%20MK3.ini

Does this mean that changes here are considered too risky, or is it worth still making suggestions/PRs?

It does seem less than optimal to heat the hot end to full target temp, then move the head all over the bed for the probe.

Other than a small piece of hard filament getting in the way, is there any other reason the hotend needs to be at target temp for levelling @rtyr?

rtyr, 2021-05-24:
It is not as simple as you may think to release breaking change like this to thousands of users. We learned from the Prusa MINI, that users have very different opinions about this (both positive and negative). Especially cooling down phase (from the preheat temperature) can be very annoying and time consuming. 

We have discussed this topic several times internally. The conclusion was, that we should have some support in the firmware first (to make it more user friendly and less annoying).  

You can always use your own custom start g-code.

trashcandrummer, 2022-08-08:
> we should have some support in the firmware first  

What exactly is meant by "some support"? Like what is the firmware missing?
Is there any timeline for this firmware change as this issue was created 4 years ago?

rtyr, 2023-02-07:
It was decided to not change the behavior on existing models (MK2/MK3). Users who prefer alternative start g-code can create separate printer profile.

> What exactly is meant by "some support"? 

Similar to Prusa MINI (it is mostly about preheat). It is also planned to use part cooling fan for cooling down phase.

closed 2023-02-07 by rtyr