prusa3d/PrusaSlicer-settings #46: PETG (and other) presets first layer cooler than other layers

triplepoint, 2018-08-31:
For several of the presets, what's the reason for running the bed and extruder first layer temperatures several degrees _cooler_ than the rest of the layers? 

It was always my understanding that, if anything, you ran the extruder and bed a bit hotter for the first layer, to promote flow and adhesion.

See, for example, "Prusa PET", though there are several similarly reversed.

Are these configurations in error, or is there a good reason for this?

triplepoint, 2018-09-05:
Any feedback here?

gyscos, 2019-02-20:
One possible reason (not sure it's what happens here) is that the first layer is slower than the others, and needs less heat to melt the filament. On other layers, faster infill may lead to a high volumetric speed, which in turn requires more heat from the hotend to maintain the melting point.

Though this would mostly apply when staying close to the melting point; I don't know if that's the case here, or if it's just a typo.

rennne, 2019-02-21:
A reason could be that PETG tends to stick a bit zu well to the PEI print bed. The hotter the worse the effect.

Another thing is that when PETG is preheated to a higher temperature during the z-calibration, the print bed gets filled with ozing and stinging. Ideal would be a seperate "nozzle pre heat temperature" setting in slic3r which would be used for that, instead of the "first layer temperature"

But generell high printing temp is  good for layer adhersion

rtyr, 2019-04-03:
Hi. Reason behind this is to reduce adhesion of PETG filament to PEI.

closed 2019-04-03 by rtyr