prusa3d/PrusaSlicer-settings #50: Filamentum CPE settings far too high

kif, 2018-09-22:
Hi,

I just purchased a spool of this Filamentum PET from Prusa (the "black soul" one) to print on the i3Mk2s printer.
https://github.com/prusa3d/Slic3r-settings/blob/60ab8871f2a7e27abe8eb639bf0e69b98d161816/live/PrusaResearch/0.2.5.ini#L818

The defaults temperature settings in Slic3rPE for this filament are FAR to HIGH. The filament bubbles all the time at 275° with tons of hair making really ugly prints. I initially believed it was "wet" filament which is unlikely for a brand new spool on the one hand and PET on the other. It turned out it was actually much to high. I got much better prints at 260-265° for the nozzle. I also reduced a bit the bed temperature to 80° to lower the adhesion which was too strong: I don't want to damage the PEI sheet.

Moreover the default settings are really at the upper limit suggested by filamentum:

Working temperature: | 255 - 275 °C
Hot bed: | 70 - 85 °C
https://fillamentum.com/collections/cpe-filaments/products/cpe-hg100-black-soul

jindrichbenes, 2018-11-14:
Settings for fillamentum CPE were adjusted by direct fillamentum request. We also did have experience with bubbles and ugly prints from CPE, but drying the spool solved this issue. It appears CPE does have tendency to produce worse results when wet.

kif, 2018-11-14:
On Wed, 14 Nov 2018 02:45:16 -0800
jindrichbenes <notifications@github.com> wrote:

> Settings for fillamentum CPE were adjusted by direct fillamentum request. We also did have experience with bubbles and ugly prints from CPE, but drying the spool solved this issue. It appears CPE does have tendency to produce worse results when wet.

My spool was completely new, freshly unpacked. I don't think water
content was the issue.
It took me a couple of days to tune the extrustion temperature and I am
now happy with 265/260°C

While I finally got decent prints, I did not find the CPE superior any
way to other PET-G I work with.

boelle, 2019-01-16:
@jindrichbenes i will risk my neck here, but would it not have been smart to verify what filamentum request before making the change? after all its prusa's good name that takes any blame for wrong settings

rtyr, 2020-06-30:
This filament is very hygroscopic and most common issues are related to that.

I would not assume that freshly unpacked filament is perfectly dry, I have different experience unfortunately.

Sealed bag with dessicant / Dried overnight / After roughly 24hours unpacked
[image]

Closing.

closed 2020-06-30 by rtyr