prusa3d/PrusaSlicer-settings #53: MMU2 Prima Select pva+ profile??

mylife4aiurr, 2018-10-09:
https://shop.prusa3d.com/en/special/169-primaselect-pva-soluble-support-500g.html
From product page:
"3D Prima's new formula for PrimaSELECT PVA+ is more stable to print with than regular PVA. This also means that it bonds much better than regular PVA to PLA, ABS and PETG.
Tested and recommended as soluble supports for our Multi Material upgrade."

Tested and recommended? - ok where's the MMU2 profile?

I also asked this question in Prusa mmu2 forums before coming here 2 weeks ago no answer.
https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f56/primaselect-pva-profile--t24561.html

rtyr, 2019-02-18:
Hi. PrimaSelect PVA+ profile for MMU2/S is available in recently released settings bundle (0.4.0/Slic3r 1.41.3).

closed 2019-02-18 by rtyr

gysnst2, 2019-07-04:
Can I reopen this im am trying M3D water soluble in the mmu2 it loads fine but the ir filament sensor does not detect the PVA and will make a big blob on the purge tower.  then it will unload the filament reload the filament re blob it on the purge block and then the bondtech gears just keep spinning

rtyr, 2019-07-04:
Hi. This issue was about missing filament profile for one specific PVA brand (PrimaSelect PVA+). Your problem is most probably hardware related (IR calibration). Check this article about IR filament sensor calibration: https://help.prusa3d.com/article/wprwemqsmj-ir-filament-sensor-calibration. PVA filaments are often very soft (like flex) which means, that idler door will open slightly less after loading the filament. Also check your idler tension. If none of this helps, please contact support via live chat at https://shop.prusa3d.com.