prusa3d/PrusaSlicer-settings #56: Mk3 profile infill is worse than Mk2S profile infill

johnbiggs, 2018-12-10:
It's not clear why but the Mk2 uses cubic and the Mk3 uses grid. This feels like a step back. It is super easy for me to change in my slicer but it seems the Mk3 default should be updated. Gyroid perhaps?

rongith, 2018-12-18:
grid is more printable than cubic. i think they want a more reliable infill. you can create your own profiles. i find gyroid quite slow and nice only for 20% infill. i always use rectilinear as i dont want crossing pathes for petg like filaments, too much build up on the nozzle.

rtyr, 2020-04-05:
MK3 is using combination of grid and gyroid now. There are some advantages and disadvantages of each fill pattern. Grid is a very universal pattern. Closing.

closed 2020-04-05 by rtyr