prusa3d/PrusaSlicer-settings #57: Add a way to disable all Prusa provided filaments for user printer

fredizzimo, 2019-01-01:
I would want to define my own list of filaments, with my own settings for my printer. Therefore it would be nice if I could exclude all the Prusa provided ones (both Prusa branded and others) from the list. 

This could be as simple as checking that a tag exist in the printer notes, for example "Disable Prusa Filaments"

boelle, 2019-01-15:
remove the checkbox that says all printers are compatible under each filaments advanced settings

then it will only be an option if you have a prusa printer

boelle, 2019-01-15:
no that will not work, my bad

so far you can be lucky that it will start with the filament you used last, but i agree that there should be a way to at least hide the defaults

bubnikv, 2019-01-15:
sorry, I dont understant. A more detailed description would help.

boelle, 2019-01-15:
He wants an option to hide all default filaments and printers

fredizzimo, 2019-01-15:
As @boelle, said, basically I would like to have a way to hide all default filaments and ideally also have the ability to create my own filament with the same name.

We can't assume that the ideal printing temperatures for example stays the same on different printers. Therefore the default list of filament is useless to me. 

One way of doing this, which would require no UI work or changes to Slic3r itself, would be to add an additional condition to all your filaments. Something that you can add to the printer notes to make them incompatible. That's why I created the issue here and not for Slic3r.

Another way of doing it, would be to have to specifically enable the filaments, like you have to do for printers, but that would need changes to the configuration wizard, and maybe to the configuration system itself.

A third way of doing it would be to completely disable the whole Prusa vendor configuration. I'm actually building my own set of profiles using the vendor system, since it's much more flexible than the normal Slic3r configuration system, so that also means that I don't really need the default print profiles either.

bubnikv, 2019-01-16:
> One way of doing this, which would require no UI work or changes to Slic3r itself, would be to add an additional condition to all your filaments. Something that you can add to the printer notes to make them incompatible. That's why I created the issue here and not for Slic3r.

We can do that. 

> A third way of doing it would be to completely disable the whole Prusa vendor configuration.

Do you own one of our printers? If not, then the simplest solution for you is to not install the Prusa3D profiles at all.

> Edit Now that I think of it, I could probably do the third way myself, by deleting the Prusa vendor files from the datadir. I think they are only copied there if the datadir is initially empty. But I have not tested it

Exactly. That will achieve the same effect as not installing the Prusa profiles at all. When running the wizard, go the "alternate" way, and the Prusa profiles should not get installed at all.

In the future, we will likely support overrides inside the filaments section, where you would be able to override any printer/print/filament preset value based on a condition made of any printer/print preset value. But that will not come soon.

boelle, 2019-01-16:
@bubnikv while we are on the subject

I have found that filaments from Devil Design can be run with the same settings as prusament, are there any rules on submitting new filaments? ie do it have to be tested on a prusa printer? i have a mendel90 with full marlin firmware but same settings as prusa firmware. i also use slic3r PE and i run my printer as if its a mk2.5, the only changes are the start and stop gcode and i use 3 perimiters and not 2

bubnikv, 2019-01-16:
Our Prusa3D vendor profiles contain filaments, which were tested by us. The
Prusa3D vendor filaments are only installed, if you install any of the
Prusa3D printers (actually there may be a bug there, that the filaments are
installed even if you disabled all the Prusa3D printers in the install
wizard).

If your filament profiles are compatible with the MK2.5, then we may be
interested adding them to our default profiles, please post them here. We
may need to validate your profiles in house first.

boelle, 2019-01-16:
i mean i use the prusament pla profile with devil design PLA, how do i post that?

bubnikv, 2019-01-16:
The prusament PLA profile is the same as the generic. The best would be to
add a note "compatible with the devil design PLA" to the notes of the
generic PLA profile.

boelle, 2019-01-16:
made a PR for it, will do the same for ABS

italocjs, 2020-01-14:
I ended up going to C:\Users\italo\AppData\Roaming\PrusaSlicer-alpha  (i'm using alpha) and deleting the "profiles" i didnt want from their respective folders, this "fixed" the same thing you had (in my case i selected another prusa printer that i dont have just to compare the settings, but i couldnt get rid of it after comparing)

rtyr, 2020-04-05:
I believe we can close this, see release notes for 2.2.0-rc here https://github.com/prusa3d/PrusaSlicer/releases.

closed 2020-04-05 by rtyr