prusa3d/PrusaSlicer-settings #69: 0.6mm nozzle printer profiles for MMU2s

mweinelt, 2019-04-20:
Hi,

we've just upgraded from the MK3 to MK3S+MMU2S and we're now missing profiles for the 0.6mm nozzle we have been using before.

Please add the 0.6mm nozzle also for MMU2S printer profiles.

This is how far we got by reproducing data from other 0.6mm profiles. One issue is, that this does not get us 5 color 0.6mm support for some reason. 

```diff
diff --git a/live/PrusaResearch/0.4.5.ini b/live/PrusaResearch/0.4.5.ini
index 1c3d921..df25c3c 100644
--- a/live/PrusaResearch/0.4.5.ini
+++ b/live/PrusaResearch/0.4.5.ini
@@ -27,12 +27,12 @@ family = MK3
 
 [printer_model:MK3SMMU2S]
 name = Original Prusa i3 MK3S MMU2S
-variants = 0.4
+variants = 0.4; 0.6^M
 family = MK3
 
 [printer_model:MK3MMU2]
 name = Original Prusa i3 MK3 MMU2
-variants = 0.4
+variants = 0.4; 0.6^M
 family = MK3
 
 [printer_model:MK2.5S]
@@ -1950,6 +1950,16 @@ default_filament_profile = Prusament PLA
 start_gcode = M107\nM115 U3.6.0 ; tell printer latest fw version\nM83  ; extruder relative mode\nM104 S[first_layer_temperature] ; set extruder temp\nM140 S[first_layer_bed_temperature] ; set bed temp\nTx\nM190 S[first_layer_bed_temperature] ; wait for bed temp\nM109 S[first_layer_temperature] ; wait for extruder temp\nG28 W ; home all without mesh bed level\nG80 ; mesh bed leveling\n\nG21 ; set units to millimeters\n\n;go outside print area\nG1 Y-3.0 F1000.0\nG1 Z0.4 F1000.0\n; select extruder\nTc\n; purge line\nG1 X55.0 F2000.0\nG1 Z0.3 F1000.0\nG92 E0.0\nG1 X240.0 E25.0  F2200.0\nG1 Y-2.0 F1000.0\nG1 X55.0 E25 F1400.0\nG1 Z0.20 F1000.0\nG1 X5.0 E4.0 F1000.0\n\nM221 S{if layer_height<0.075}100{else}95{endif}\nG90 ; use absolute coordinates\nM83 ; use relative distances for extrusion\nG92 E0.0\n
 end_gcode = G1 X0 Y210 F7200\nG1 E2 F5000\nG1 E2 F5500\nG1 E2 F6000\nG1 E-15.0000 F5800\nG1 E-20.0000 F5500\nG1 E10.0000 F3000\nG1 E-10.0000 F3100\nG1 E10.0000 F3150\nG1 E-10.0000 F3250\nG1 E10.0000 F3300\n\nM702 C\n\nG4 ; wait\nM104 S0 ; turn off temperature\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\nG1 X0 Y200; home X axis\nM84 ; disable motors
 
+## MODIFICATION BEGIN^M
+[printer:Original Prusa i3 MK3S MMU2S Single 0.6 nozzle]^M
+inherits = Original Prusa i3 MK3S MMU2S Single^M
+nozzle_diameter = 0.6^M
+max_layer_height = 0.40^M
+min_layer_height = 0.15^M
+printer_variant = 0.6^M
+default_print_profile = 0.30mm QUALITY 0.6 nozzle MK3^M
+## MODIFICATION END^M
+^M
 [printer:Original Prusa i3 MK3S MMU2S]
 inherits = *mm2s*
 machine_max_acceleration_e = 8000,8000
@@ -1958,6 +1968,16 @@ extruder_colour = #FF8000;#DB5182;#00FFFF;#FF4F4F;#9FFF9F
 start_gcode = M107\nM115 U3.6.0 ; tell printer latest fw version\nM83  ; extruder relative mode\nM104 S[first_layer_temperature] ; set extruder temp\nM140 S[first_layer_bed_temperature] ; set bed temp\nM190 S[first_layer_bed_temperature] ; wait for bed temp\nM109 S[first_layer_temperature] ; wait for extruder temp\nG28 W ; home all without mesh bed level\nG80 ; mesh bed leveling\nG21 ; set units to millimeters\n\n; Send the filament type to the MMU2.0 unit.\n; E stands for extruder number, F stands for filament type (0: default; 1:flex; 2: PVA)\nM403 E0 F{"" + ((filament_type[0]=="FLEX") ? 1 : ((filament_type[0]=="PVA") ? 2 : 0))}\nM403 E1 F{"" + ((filament_type[1]=="FLEX") ? 1 : ((filament_type[1]=="PVA") ? 2 : 0))}\nM403 E2 F{"" + ((filament_type[2]=="FLEX") ? 1 : ((filament_type[2]=="PVA") ? 2 : 0))}\nM403 E3 F{"" + ((filament_type[3]=="FLEX") ? 1 : ((filament_type[3]=="PVA") ? 2 : 0))}\nM403 E4 F{"" + ((filament_type[4]=="FLEX") ? 1 : ((filament_type[4]=="PVA") ? 2 : 0))}\n\n{if not has_single_extruder_multi_material_priming}\n;go outside print area\nG1 Y-3.0 F1000.0\nG1 Z0.4 F1000.0\n; select extruder\nT[initial_tool]\n; initial load\nG1 X55.0 E29.0 F1073.0\nG1 X5.0 E29.0 F1800.0\nG1 X55.0 E8.0 F2000.0\nG1 Z0.3 F1000.0\nG92 E0.0\nG1 X240.0 E25.0  F2200.0\nG1 Y-2.0 F1000.0\nG1 X55.0 E25 F1400.0\nG1 Z0.20 F1000.0\nG1 X5.0 E4.0 F1000.0\nG92 E0.0\n{endif}\n\nM221 S{if layer_height<0.075}100{else}95{endif}\nG90 ; use absolute coordinates\nM83 ; use relative distances for extrusion\nG92 E0.0\n
 end_gcode = {if has_wipe_tower}\nG1 E-15.0000 F3000\n{else}\nG1 X0 Y210 F7200\nG1 E2 F5000\nG1 E2 F5500\nG1 E2 F6000\nG1 E-15.0000 F5800\nG1 E-20.0000 F5500\nG1 E10.0000 F3000\nG1 E-10.0000 F3100\nG1 E10.0000 F3150\nG1 E-10.0000 F3250\nG1 E10.0000 F3300\n{endif}\n\n; Unload filament\nM702 C\n\nG4 ; wait\nM104 S0 ; turn off temperature\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\n; Lift print head a bit\n{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+30, max_print_height)}{endif} ; Move print head up\nG1 X0 Y200; home X axis\nM84 ; disable motors\n
 
+## MODIFICATION BEGIN^M
+[printer:Original Prusa i3 MK3S MMU2S 0.6 nozzle]^M
+inherits = Original Prusa i3 MK3S MMU2S^M
+nozzle_diameter = 0.6;0.6;0.6;0.6;0.6^M
+max_layer_height = 0.40^M
+min_layer_height = 0.15^M
+printer_variant = 0.6^M
+default_print_profile = 0.30mm QUALITY 0.6 nozzle MK3^M
+## MODIFICATION END^M
+^M
 # The obsolete presets will be removed when upgrading from the legacy configuration structure (up to Slic3r 1.39.2) to 1.40.0 and newer.
 [obsolete_presets]
 print="0.05mm DETAIL 0.25 nozzle";"0.05mm DETAIL MK3";"0.05mm DETAIL";"0.20mm NORMAL MK3";"0.35mm FAST MK3";"print:0.15mm OPTIMAL MK3 MMU2";"print:0.20mm FAST MK3 MMU2"
```

rtyr, 2019-04-24:
Hi. It is planned, but it will be most likely available only for Slic3rPE 1.42.x and above (Beta2 at the moment). Resulting printer profile names are unfortunately very long and they are not displayed correctly in some cases with current stable Slic3rPE 1.41.3.

rtyr, 2019-05-07:
0.6mm and 0.25mm MMU2 printer profiles are now available in latest settings bundle (0.8.0-beta4 and above) for Slic3rPE 1.42.0 (beta). Printer profiles can be activated through configuration wizard. Only for single mode for now.

3d-gussner, 2019-07-07:
If that is fixed why not close the issue?

mweinelt, 2019-07-07:
Because it only works for single filament prints, but not multi-filament prints.

rtyr, 2019-08-13:
MMU profiles for 0.6mm nozzle are now available in PrusaSlicer 2.1.0-alpha1, so you can try it :-).

rtyr, 2019-09-17:
0.6mm nozzle print profiles are now included in latest PrusaSlicer 2.1.0. Closing.

closed 2019-09-17 by rtyr