prusa3d/PrusaSlicer-settings #70: [Request] Define location for user saved presets

mvasilakis, 2019-05-20:
It would be nice to be able to define the save location for user saved presets. For OS X they are saved in /users/user/Library/Application Support/Prusa Slicer/(print, printer, filament)

While it's fairly trivial to create a hard link and place the directory where I'd like it might be useful for others to have the ability to define this location. My reason for this is so that I can define a dropbox directory where my profiles reside. This way I have access to them on any computer i use and they are the latest saved versions.

bubnikv, 2019-05-21:
There is a command line parameter --datadir that does what you need.

bubnikv, 2019-05-21:
Does it help?

mvasilakis, 2019-05-21:
Yes and no. 
This command does exactly what I'm looking for. Unfortunately the app must be started via command line every time otherwise user profiles default to ~/Library/Application\ Support/PrusaSlicer. A preference that sets and saves this would be nice.

Command used:
/Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --datadir /Volumes/MacOS/Users/me/Dropbox/3D\ Printer/PrusaSlicer\ App\ Profiles

OS - Mac OS X 10.14.4 Prusa Slicer 2.0 final.

vojtechkral, 2019-05-21:
> A preference that sets and saves this would be nice.

_Where_ would you save this preference? That seems to be a sort of a chicken and egg problem :)

mvasilakis, 2019-05-21:
Configuration/Preferences in the GUI? Possibly as part of a plist in ~/Library/Preferences/com.prusa3d.slic3r

mvasilakis, 2019-05-21:
> Couldn't you solve your problem with a symlink instead?
I did. After bubnikv told me about --datadir I wrote a little script to use as a launcher as well. Not very elegant and I need to remember to launch via script otherwise I wont have the latest profiles. 
Plus I'm more of a power user that most.

mvasilakis, 2019-05-21:
If anyone is interested I wrote a little script and wrapped it to an [app](https://www.dropbox.com/s/31ksde5pz1mh4hn/PrusaSlicer%20Launcher.zip?dl=0) using [platypus](https://www.macupdate.com/app/mac/12046/platypus)

The script creates a directory in ~/Dropbox called "PrusaSlicer App Profiles" and then creates a directory called _Backups within that. Every time it's launched it creates a zip backup of the current profiles and placed them in the _Backups directory. The last 25 backups are kept and named by timestamp. 

PrusaSlicer is then launched using this directory for your profiles. 

Notes: PrusaSlicer needs to reside in /Applications directory and Dropbox needs to be in the root user directory. ( ~/Dropbox). If either is not then the script can be modified (R click "PrusaSlicer Launcher.app", show package contents. Then navigate to: contents, resources, script. OR add an alias where expected. 

Hope this helps someone. It's by far not an ideal solution. (Ideal would be a preference within the app.) but I've been bouncing from computer to laptop and it's frustrating to send a print to the printer only to remember you made a change that's not in the current slice.

Sembazuru, 2019-09-19:
I would also like this for use on Windows machines. The --datadir "solution" doesn't really work for me as I may start PrusaSlicer from an icon on the desktop, from the icon in the start menu, or by pressing the windows key and starting to type PrusaSlicer and pressing enter when the search results pull up PrusaSlicer for me.
I'm not sure how to add command line arguments to any of those three launching methods, having a user-presets setting would just be simpler for me in the end.

bubnikv, 2019-09-19:
1) Create shortcut 
2) Open shortcut properties, add the --datadir parameter

[image]

Breuxi, 2019-10-12:
Found [this](https://manual.slic3r.org/configuration-organization/configuration-organization) in the Slic3r Manual, could maybe also help for PrusaSlicer :)
> On Linux and MacOS X you have another option instead of using --datadir: a link to the shared directory can be created where Slic3r looks for the default directory.  
This can used for example for synchronizing Slic3r presets using Dropbox.  
The first step is to find the path where Slic3r looks for its configuration:

```
$ /Applications/Slic3r.app/Contents/MacOS/slic3r --debug
wxWidgets version wxWidgets 2.9.4, Wx version 0.9922
Data directory: /Users/al/Library/Application Support/Slic3r
```

> Now, after closing Slic3r, we move this directory to our shared folder:
`mv "/Users/al/Library/Application Support/Slic3r" ~/Dropbox/Slic3r-settings`
(Double quotes are needed because of the space in the path, but if you're doing this you should already know how to work with the terminal.)
And finally we create a symbolic link:
`ln -s ~/Dropbox/Slic3r-settings "/Users/al/Library/Application Support/Slic3r"`

eisenhowerj, 2019-11-20:
Would be nice if configuration storage worked with git itself. I'll be setting up a way to do it on my local machine but that will require me to do something outside of Slicer every time I want to commit. This feature should work with branching e.g. if I want to start playing with settings, I would go into Slicer settings and update the application config to use a new branch.

mvasilakis, 2019-11-20:
> Would be nice if configuration storage worked with git itself. I'll be setting up a way to do it on my local machine but that will require me to do something outside of Slicer every time I want to [...]

bubnikv and I discussed this in a #924. 
https://github.com/prusa3d/PrusaSlicer/issues/934

I haven't had the time to mess with it. I gotta say though it doesn't look quite that complicated. I see no reason that the git couldn't be used to store profiles. This would also be a great way for a resin co to get their SL1 profiles out there and current. 

As it stands I use the launch script I linked above to set my --datadir to my dropbox. It's been working fine so I haven't had a pressing need to use vojtechkral's notes.

eisenhowerj, 2019-11-21:
I dont use Dropbox so I'm not familiar; does it support file versions? What happens if things were working last week and you want to revert to that config?

This is what I'll be doing:
```shell
# Initialize where git will store its data, setup an alias
git init --bare $HOME/.PrusaGit
alias slicerconfig='/usr/bin/git --git-dir=$HOME/.PrusaGit/ --work-tree=$HOME/.PrusaSlicer'
echo "alias slicerconfig='/usr/bin/git --git-dir=$HOME/.PrusaGit/ --work-tree=$HOME/.PrusaSlicer'" >> .bash_profile
slicerconfig config --local status.showUntrackedFiles no
cd .PrusaSlicer/
vi .gitignore # ignore cache folder
slicerconfig add . # I don't recall if this has to be inside your .PruseSlicer directory
slicerconfig status
slicerconfig remote add origin git@gitlab.com:eisenhowerj/slicerconfig.git
slicerconfig commit -m "initial commit"
slicerconfig push -u origin master
```
Now I can do things such as:
```shell
$ slicerconfig checkout -b change-filament-temps
```
Change some settings in the UI
```shell
$ slicerconfig status
On branch change-filament-temps
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   filament/MP Mini Delta.ini

no changes added to commit (use "git add" and/or "git commit -a")
```
```shell
$ slicerconfig diff -U0
diff --git a/filament/MP Mini Delta.ini b/filament/MP Mini Delta.ini
index 20acada..01ff4e3 100644
--- a/filament/MP Mini Delta.ini        
+++ b/filament/MP Mini Delta.ini        
@@ -1 +1 @@
-# generated by PrusaSlicer 2.1.0 on 2019-11-21 at 00:01:14 UTC
+# generated by PrusaSlicer 2.1.0 on 2019-11-21 at 00:11:12 UTC
@@ -8 +8 @@ compatible_prints_condition =
-cooling = 1
+cooling = 0
@@ -12 +12 @@ extrusion_multiplier = 1
-fan_always_on = 0
+fan_always_on = 1
@@ -14 +14 @@ fan_below_layer_time = 60
-filament_colour = #29B2B2
+filament_colour = #EF2929
@@ -46,2 +46,2 @@ filament_wipe = nil
-first_layer_bed_temperature = 60
-first_layer_temperature = 210
+first_layer_bed_temperature = 55
+first_layer_temperature = 205
```
```shell
$ slicerconfig add . # NOTE: This works, even if you're not in the .PrusaSlicer config directory since the alias has --working-dir=$HOME/.PrusaSlicer
$ slicerconfig commit -m "testing changes in temps"
[change-filament-temps ccf2fc2] testing changes in temps
 Committer: johnnye <johnnye@lappytop.lan>
 1 file changed, 6 insertions(+), 6 deletions(-)
$ slicerconfig push -u origin change-filament-temps
Enumerating objects: 7, done.
Counting objects: 100% (7/7), done.
Delta compression using up to 8 threads
Compressing objects: 100% (3/3), done.
Writing objects: 100% (4/4), 411 bytes | 411.00 KiB/s, done.
Total 4 (delta 2), reused 0 (delta 0)
remote: 
remote: To create a merge request for change-filament-temps, visit:
remote:   https://gitlab.com/eisenhowerj/slicerconfig/merge_requests/new?merge_request%5Bsource_branch%5D=change-filament-temps
remote: 
To gitlab.com:eisenhowerj/slicerconfig.git
   92a4f94..ccf2fc2  change-filament-temps -> change-filament-temps
Branch 'change-filament-temps' set up to track remote branch 'change-filament-temps' from 'origin'.
```

ideaalab, 2020-07-15:
Git may be a bit to much for a standard user... Just a user defined path to save user profiles and configs would be just great. You can have all PrusaSlicer in sync on different computers, using Dropbox, OneDrive, GoogleDrive, CreativeCloud, SynologyDrive or any of the many others...