prusa3d/PrusaSlicer-settings #71: [Suggestion] Output Filename

cycl0ne, 2019-06-05:
You lately changed the filename output to the time being used. Thats wonderfull and this took me to dig more into the filename thing. So my request is as follow, could you change the default output name to:
{input_filename_base}{layer_height}mm{filament_settings_id[0]}{printer_model}{print_time}.gcode

filament_settings_id instead of filament_name?

You may ask why? Easy: One has different PLA which is used, with different temperature. If the output file is just:

name_0.2mm_PLA_MK3S_27m.gcode

its not usable, but when used the filament_settings_id:

name_0.2mm_PRUSA_PLA_MK3S_27m.gcode
name_0.2mm_GENERIC_PLA_MK3S_27m.gcode

its makes more sense.

rtyr, 2019-06-06:
Hi. I am affraid that output filename would be too long with filament_settings_id. But you can use your own custom filament type now. 

[image]

rtyr, 2019-07-17:
Can I close this issue?

cycl0ne, 2019-07-17:
Yes, of course.

closed 2019-07-17 by cycl0ne