prusa3d/PrusaSlicer-settings #74: Decrease min_print_speed

fornellas, 2019-07-04:
This value is currently set to 15mm/s, for all filaments. I noticed that I've been getting poor performance in layers with small area (eg: a few mm^2), and traced the issue to this parameter.

At [some point in time](https://github.com/prusa3d/PrusaSlicer-settings/blob/e34d758a36d50581cebf7b76378613eaeeda6433/Slic3r%20settings%20175/filament/Prusa%20PLA%201.75mm.ini#L15), this was set to 5mm/s, didn't find any commit log explaining the rational to increase it to 15mm/s.

I did some testing using Prusament PLA at my MK2.5, 0.15mm layer, 0% infill. The test object was a pointy cone 30mm tall. With the default settings (15mm/s), the last 5mm or so of the cone, have signs of poor cooling / overheating, similar to what is described [here](https://www.simplify3d.com/support/print-quality-troubleshooting/overheating/) (but not as bad as that picture). I then set min_print_speed to 1mm/s, and... got a 100% perfect result!

Not sure if we really need to go as slow as 1mm/s, perhaps 5mm/s would suffice (would require more testing), but it seems clear the the current value of 15mm/s is just too high to get decent results with small area layers.

So... can we have the default value decreased, so we can get working prints out of the bok?

rtyr, 2019-07-15:
E3D V6 all-metal hotend needs some filament flow to work reliably, especially when printing PLA. Extremely slow printing speeds significantly increases risk of hotend clog caused by heat creep.  There are also some other issues (different appearance/glossines, increased printing times, ...). Better solution for printing models like this is to print more copies at once (or print it together with some other model with appropriate height).

fornellas, 2019-07-17:
Hummm.. TBH, I already get different gloss due to fan on the first few layers with PETG from Prusa, not sure how big of a deal it is. Having malformed prints being better than gloss change is a hard sell... 

I didn't mention, but I had similar issues with ABS and a roughly 1cm2 area tower (bigger than the PLA test I shared). I only managed to print without issues by decreasing this value. From these 2 data points, it seems different materials would need / have different thresholds and correctly 15mm/s applies to 100% of filaments.

I don't have enough data points to argue what's the threshold for no clogging, but it would be nice to take an educated choice on this value.

fornellas, 2019-07-21:
For those interested, I did some more testing with PLA.

# First test: 1.9mm diameter tube, 15mm tall

Making this extreme case work, should make things better overall. I printed with a single perimeter, at several speeds:

- 15mm/s: malformed.
- 7mm/s, 3mm/s: well formed, but print is 1.75mm diameter (not 1.9).
- 1mm/s: well formed, 1.84mm diameter (close enough).
- 0.3mm/s: malformed.

So it seems, 1mm/s is the sweet spot.

# Second Test: 41mm diameter cone, 34mm tall.

Having 1mm/s as a target, we can also tweak "Slow down if layer print time is below Xs" to hit that threshold only when needed. Again, more tests, printed with with no top/bottom solid layers and 0% infill:

- Default settings:
  - Slow down if layer print time is below: 20s
  - Min print speed: 15mm/s
  - Print time: 25m35s.
  - Quality: OK
- Tweaked settings:
  - Slow down if layer print time is below: 6s
  - Min print speed: 1mm/s.
  - Print time: 18m.
  - Quality: OK

The default settings print the whole cone at 15mm/s. The tweaked settings lets it go at 50mm/s at the base, and only slow down to 1mm/s at the very top, making it ~40% faster.

# Results

From the experiments, the default settings are both slower and deliver worse quality than the tweaked settings:

- Slow down if layer print time is below: 6s
- Min print speed: 1mm/s.

Across all my testings, I experienced no clogging due to heat creep, though this has pointed to be an thing for speeds lower than 15mm/s.

If we keep the min print speed at 15mm/s, the results suggest that lowering "Slow down if layer print time is below 20s" to 6s should have no impact in print quality, but delivers significant gains in print speed. I thus suggest setting this for PLA at least, as testing with other materials have to be done to confirm benefits.
[image]
[image]
[image]
[image]
[image]

jonnyhifi, 2019-07-21:
What a superb post you've put together here. This I saw through your facebook post which brought me here.
I've been trying to dial in Nylon (Taulman alloy 910) and was suffering terrible stringing on my temp tower .  
I'd thought I needed to dial in retraction settings, but only after quite a bit of trial and error did I similarly hit upon these parameters which were confounding my testing, as the stringing test was with a very small component. (measuring 1cm or so across). Lots of the strings towards the tops of the classic stringing test piece (which were two points sticking up), were particularly terrible, and this was because the nylon hadn't solidified enough. Your methodology gives a nice way to go about optimising these parameters in an objective way. I tried one experiment setting minimum speed to zero and layer time to 24 seconds as I remember : and the downside was towards the tops of the points : the nozzle was barely moving and it seemed to be oxidising the nylon by continual heating  (it turned a brown colour from its normal neutral white), so another limit on how slow one can let printing get, may be degradation of the filament, not just jams. (My mk3 has a bear extruder with a sunon fan and titanium heatbreak, so touch wood I would have an extruder highly unlikely to jam through heat creep). It will be interesting to see what sort of parameters here would suite nylon, and whether they are largely inline with what you've found for PLA. Certainly the stock "15 mm/s" essentially prevents useful slowing down for small components. This is of particular interest to me as I wish to print some very small functional components in Nylon, hence really need to understand this issue. Thank you once again!

closed 2023-07-14 by rtyr