prusa3d/PrusaSlicer-settings #75: Number of layers for thin objects incorrect.

AbeFM, 2019-07-15:
2.0.0 win10 mk3smmu2s
I wanted to trigger a filament at a few places, so I added several cylinders into the field, and set them to the desired extruder, all other objects set to main extruder.. I lined them up so I would have a single tower. To make that easier (their x-y values didn't correspond, see next paragraph) I made each one slightly smaller diameter. See next paragraph.

Trying to position them, I found they'd inherited their coordinate system from the imported object's system. It would be great if I could generate them in the environment directly in addition to tied to the object.

Anyway, the movement arrows when you click in the X/Y/Z Position Part Manipulation box aren't square to the printer's system, and while you can move them in x/y/z with the arrow tool, the z-values displayed are the projection, so to move 10 mm you can't add 10 mm to the Z value.

This leads to another problem, I think - I'm printing at .15mm layer heights. I set these objects to 0.15 mm and one worked fine. 2 didn't get sliced. I increased to 0.2, 0.3, 0.35 before it would render. At 0.4, it would slice 2 layers thick even though the thickness of the object differ by only 1/3 of a layer thickness.

The 3rd object was even worse, not only did I need to set the z "size" at 0.375 to get it to render 1 line (0.37 wouldn't appear and there'd be no support for it either). When I set it to 0.38 I would get 2 layers thick. You have to type in 3.75 (which, confusingly, renders as 0.38 in the box, after you type it you can see it change, but the real value is there as you can tell the difference in the slicing even with the same displayed value....).

So, I think the numbers are misleading in the box, I think they aren't being printed unless their projection in the printer's Z-axis is thick enough, and so the slope they cross that threshold at depends on the diameter of the discs.

Model:
[image]
You can see my 3 decreasing diameter discs stacked by eye on a vertical center.

Desired Tower:
[image]
Discs sliced as desired, each rendered one layer thick with the main material used for supports.

Resultant tower if I use 0.05 thicker part (with 0.15 thick layers)
[image]

rtyr, 2019-09-17:
I believe this issue should be posted at https://github.com/prusa3d/PrusaSlicer. Can you please move it there? Preferably with 3MF project file included.

AbeFM, 2019-09-18:
Thanks for that! Too much stuff dies because it's not "my problem".

closed 2019-09-18 by AbeFM