prusa3d/PrusaSlicer-settings #78: Feature Request: Start Gcode Enhancement

malcinator, 2019-09-17:
### Version

Prusaslicer 2.1.0 Combined with Firmware 3.8.0

### Operating system type + version

macOS 10.14.6 or any

3D printer brand / version + firmware version (if known)

Prusa i3 MK3S
Prusa i3 MK2.5S

### Enhancement

Add these to Prusaslicer config ini files

; Prusaslicer Settings Used
; Printer Settings - [printer_settings_id]
; Filament Settings - [filament_preset]
; Print Settings - [print_settings_id]

I have two Prusa Printers & and couple of other brands, with the arrival of 2.1.0 Prusaslicer & firmware 3.8.0 I see that there is a check for the correct sliced code being printed on the matched printer with M862.3 P "[printer_model]". I use different profiles for almost every type of filament, Print Settings & Custom Printer Profiles and combining this with some simple automation via a mapped share & Octoprint 'Watched folder', the M862 is getting in the way of my workflow, I could for example print a sliced MK2.5S file on a MK3S and vice versa, now I get the M862.3 beeping at me. Currently I can't see a way of finding out what printer & profiles I had used to slicer with, so if I send the wrong file to the wrong printer I have to re-slice it and this adds some time to my workflow.

So my question is would it be possible to add something like the above to the Prusaslicer ini files so it would be possible to see in the code which printer & profiles I had selected?

Is this a new feature request?
Yes

rtyr, 2019-09-17:
Hi. I am not sure if I understand correctly, but information about settings used, filament profile used, etc. is already included in every gcode exported from PrusaSlicer (at the end of the gcode). You can also disable preliminary checks on your printers if you don't want them (you can choose from 3 options: strict, warn, none).

Example from PrusaSlicer generated gcode:
; print_settings_id = 0.20mm SPEED MK3
; printer_model = MK3S
; filament_settings_id = "Prusament PETG"

rtyr, 2019-09-17:
I will have to discuss this with @bubnikv as the top of the gcode file is generated automatically by PrusaSlicer.

rtyr, 2023-11-21:
I am closing this issue, we are not planning to move the comments.

closed 2023-11-21 by rtyr