prusa3d/PrusaSlicer-settings #79: libnest2d googletest download during build

pmjdebruijn, 2019-09-18:
Currently libnest2d googletest unittests are forced ON during a PrusaSlicer build.

On Ubuntu 18.04 (and I suppose this will likely affect other distros as well), libgtest will no longer be autodetected by build systems, due to:
groups.google.com/group/googletestframework/browse_thread/thread/668eff1cebf5309d

This means the libnest2d build system included in PrusaSlicer will try to git clone googletest live during the PrusaSlicer build process, which shouldn't really happen. A program build should ideally never depend on any kind of network access, so it can be performed on a build farm that potentially may be highly firewalled.

So please consider including a pre-cloned copy of googletest in your source tree, so the tests can be run without having to perform a live git clone.

tamasmeszaros, 2019-09-19:
Hello and thank you for the feedback. 

The build system does not depend strongly on network access. The download is initiated only if the tests are enabled and cmake find_package is unable to find gtest, as a last resort to continue the build. You can install gtest on your build server and no download will be performed or you can disable the tests to make the configure step succeed.

We are also considering the change of gtest to another framework which would be easier to manage and does not need to be compiled separately. This decision has not been made yet.

tamasmeszaros, 2019-10-11:
Update: gtest is now replaced with catch2 test framework and there is no automatic download step in PrusaSlicer code anymore. The compilation flags issue is gone by catch2 being a header only library.