prusa3d/PrusaSlicer-settings #80: Workaround for the "Z-leveling enforced" issue

Kachidoki2807, 2019-09-26:
Hi,

I am posting there a workaround suggestion for [this issue](https://github.com/prusa3d/Prusa-Firmware/issues/1651#issuecomment-535163940) from the firmware side. I know the root cause is an electronic issue, and it can (should) be addressed by the firmware, but the quickfix workaround is easy and can benefit to everyone with no delay.

**As a reminder:**
The base issue is that the new PWM frequency of the heatbed generate an Electro Magnetic Field that seems to interfere with the PINDA, and thus make the measurement not reliable. Even is the Mesh Bed Leveling go up to the end (with or without retry), the measurement is supposed less accurate that it can be. Thus I guess <ins>the first layer quality is impacted</ins>.

**My proposal:**
To quickfix that, a slight modification of the Start G-Code can be done to _shutdown the heatbed when the PINDA is used_, so before G28 and G80, and re-enable it immediately after. The heatbed will loose maybe 1 ou 2°C during the longest MBL.

By reversing the order of waiting (extruder first and bed after) we could even benefit of the bed overshoot to save time for re-heating after the MBL, but I think it is too machine dependent.

Thanks for reading.

mrmoo06, 2019-09-27:
I am about to try this.  I shall report back.....

Kachidoki2807, 2019-09-30:
I have done some tests this weekend. We can forget this workaround. It works well enough for stuck users like me, whom wants to use it, waiting for a real solution, but it is unacceptable for a production profile.
While using PLA (60°C heatbed) it is OK because the bed loose only 2°C (at 22°C ambiant temp), but while using PETG (85°C heatbed) the bed loose more than 5°C. Not only that take additional time to reheat the bed, but also I think the thermal expansion would be different and it maybe impact the MBL accuracy. It is definitely an issue that should be fixed on the firmware side.

I let the issue open while there is no other option to solve the "Z-leveling enforced" issue

BTW, on a woobly machine (1kg brand new spool at the top of the frame), the slow retry do its job properly, whatever the PWM state.

closed 2019-09-30 by bubnikv

Kachidoki2807, 2019-09-30:
You are absolutely right. That's what I have pointed out.

bubnikv, 2019-09-30:
I have notified the firmware team to investigate your hypothesis.

ekobres, 2019-12-31:
> I have done some tests this weekend. We can forget this workaround. It works well enough for stuck users like me, whom wants to use it, waiting for a real solution, but it is unacceptable for a [...]

How does slow retry work? For now I'm stuck with the GCode fix since this issue causes Z leveling enforced about 70% of the time on my MK2.5s.