prusa3d/PrusaSlicer-settings #81: Bad supports => Feature request: Allow "sychronize with object layers"

haleypearse, 2019-10-02:
[image] _The difference between support synchronized with object layers and not. (rescued with hot glue)_

In the absence of interface layers, supports are generated at 0.298mm layer height even though my extrusion width is only 0.28mm. This activates the minimum layer time cooling, bumping the fan to full speed and slowing feedrate to a crawl. The result is no layer adhesion between the corners. 

I have resorted to some gcode:
;AFTER_LAYER_CHANGE
M106 S51 ;turn down the damn fan for supports
But the layer height is still a problem.

rtyr, 2019-10-03:
Check layer height limits in your printer profile (Printer settings->Extruder). These limits are used to cap the support layer height and variable layer height.

haleypearse, 2019-10-04:
Thanks, that works to reduce the support layer height. But even setting both min and max layer height limits equal to the part layer height, the layers are still out of sync and the cooling thresholds still apply. Perhaps the cooling thresholds should be calculated not by layer time but layer frequency.

rtyr, 2019-10-04:
I believe it is already posted in PrusaSlicer repository.  https://github.com/prusa3d/PrusaSlicer/issues/984. Closing here.

closed 2019-10-04 by rtyr