prusa3d/PrusaSlicer-settings #84: Prusament PETG suggests 250° but profile prints at 230

PxT, 2019-12-06:
Prusament PETG suggests a print temp of 250°C [here](https://shop.prusa3d.com/en/prusament/1006-prusament-petg-galaxy-black-1kg.html) but the MK3S+MMU2S profile in Prusa Slicer is only printing at 230 when the Prusament PETG setting is chosen.  Please tell me if I'm missing something.

PxT, 2019-12-06:
I think the filament definition comes from https://github.com/prusa3d/PrusaSlicer-settings/blob/fe3629e885102c7f9e3ead9b0a2aedb588b28f32/live/PrusaResearch/1.0.7.ini#L2018

rtyr, 2019-12-06:
Hi. It is by design. We are using lower temperatures for some multimaterial filament profiles (PLA, PETG) for improved reliability. It is used only with MMU2 printer profiles, "MMU Single" printer profile is using default profiles with standard temperatures. Closing.

closed 2019-12-06 by rtyr

BWBNexus, 2020-03-05:
This has bothered me for some time.  Where the temps modified downward for printing mixed PLA/PETG multiple colors?  I still have to modify my temps to successfully print multi color all PETG.  The cooler temps cause my prints to basically tear/shred.  The only difference my Prusa i3 Mk3s MMU2s has from stock is the Olsson Ruby Nozzle - High Temperature - 1.75mm x 0.40mm

PxT, 2020-03-05:
Agreed. I think this should be reconsidered, @rtyr. I have *worse* reliability at 230.

Flav1u, 2020-03-05:
So Prusa agrees that lowering the temp for PETG improves reliability for MMU2 but doesn't give us the option to set two temps, one for ramming (creating the tips) and one for printing.

DoctorMarshall, 2021-12-23:
this is strange indeed, could you elaborate on the 'reliability'?

rtyr, 2021-12-23:
> this is strange indeed, could you elaborate on the 'reliability'?

Lower temperature, less stringing, better filament tips. 

You can always create a new profile with higher temperature, if you prefer it.

rtyr, 2021-12-23:
> So Prusa agrees that lowering the temp for PETG improves reliability for MMU2 but doesn't give us the option to set two temps, one for ramming (creating the tips) and one for printing.

It doesn't make much sense, unless you wait for quite long time before the ramming :-/.  The actual temp doesn't change instantly in one second.