prusa3d/PrusaSlicer-settings #86: ABS fan speed at 15% is too low.

rongith, 2020-02-29:
Printers: Prusa MK3S

In my point of view it is an issue to lower the fan speed to 15%.
I made some tests on some of my printers.
Usually, 16% is the minimum for the fan to kick in and there is some kind of uncertainty. At 15% there is high probability the fan doesn't start at all.
To be sure, I round the speed to 20%. I could lower it to 18% but that is little optimisation.

Perhaps you have some kind of different PWM management for the fan speed on your firmware.

rtyr, 2020-03-01:
I was unable to reproduce it on stock MK3S/MK2S/MINI (minimum for MK3/MK2 was around 7%). What firmware are you using?

I will check it further on more printers.

rongith, 2020-03-01:
I rechecked and my published data seems old.
As you, my fan starts now as low as 16/255. And I noticed there is is a boost at the start that didn't exist before (since firmware version?). Then with the help of the boost the fan start too at 15/255 but it is not enough to maintain the fan running.
Low fan speed may not be good for all firmwares.
I will keep your settings.

closed 2020-03-01 by rongith