prusa3d/PrusaSlicer-settings #88: System material presets all gone but one 1.1.2

6ffm70, 2020-03-27:
SW Version PrusaSlicer 2.2.0+linux-x64
Settings version 1.1.2
Printers configured Mk3s and MINI

**Problem Description**
Only saved user presets appear in both filament selection menus with the exception of one system preset: Prusament PETG@Mini.

**Reproduction**
Downgrading (overwrite $PATH/vendor/PrusaResearch.ini) to Settings version 1.1.1 restores all system presets. Upgrading in the same way reverts as expected.

rtyr, 2020-03-27:
It should be possible to enable missing filament profiles in configuration wizard. Does it work?

6ffm70, 2020-03-30:
Yes, that does work!
Yet, this method is not very obvious when updating the application; ie. I only now found a (an indirect) hint in the release notes from  2.2.0-alpha2, as I only use stable releases instead of the admittedly impressive number of pre-releases.

rtyr, 2020-03-30:
All filament and material profiles from previous PrusaSlicer version should be automatically enabled in the 2.2.0 final. It didn't happen in your case for some reason.

6ffm70, 2020-03-31:
This might possibly have to do with the combination of using an appimage and/or using the command line option --datadir=

6ffm70, 2020-04-02:
Certainly no issue of the Settings-release missing entriees.

closed 2020-04-02 by 6ffm70