prusa3d/PrusaSlicer-settings #94: Filament profiles do not follow FW 3.9.0 recommendations

fiveangle, 2020-06-03:
Per [3.9.0 release notes](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0), the following issues are present in current filament settings:

1) LA is not reset at end of print:
> In general, the M900 K0 resets the LA and it is recommended to have it at the end of every G-code.

2) LA 1.5\nLA 1.0 value pair lines are not present:

> A special note regarding the calibration prints created under LA 1.0. The automatic conversion to the LA 1.5 might not bring the same print results compared to the previous version. This is because the automatically converted K factors (LA 1.0 to LA 1.5) can differ from the ideal K factors for LA 1.5. It is highly recommended to reslice the calibration object having the LA 1.5 values already in the G-code.
> 
> **In general, it is recommended that the user switches to the new [LA 1.5 value] profiles whenever possible.**

Auto-conversion from LA 1.0 to LA 1.5 format by the FW beginning with 3.9.0 occurs, but as per 2 above, it seems it would be smarter to include the pairs in the profiles so users will understand the parameters that should be changed rather than mistakenly tune based on converted LA1.0 values.

fiveangle, 2020-06-03:
Immediately prior to testing the 2 points above I launched PS 2.2.0 release and it presented the profile update dialog which I accepted.  At that point, I assumed I had the latest profiles and tested the above points then immediately filed the issue here.

However, based on your request I checked and profiles 1.1.3 were loaded.  I then quit PS and relaunched and was then presented again with the profile update dialog showing 1.1.6 update available.  I accepted and confirmed both items above are present in the resulting GCODE so will close this issue.

HOWEVER: Is this behavior of PS working as designed, or is this a bug that should be filed against the PS branch ?  The previous launch that updated to 1.1.3 should have updated to 1.1.6, correct ?

rtyr, 2020-06-03:
Behavior you are describing happens when you close update dialog without either confirming or declining the update. It can happen quite easily, so I would assume that something like that probably happened.

fiveangle, 2020-06-03:
I attempted to manually restore the 1.1.2 -> 1.1.3 snapshot to re-test the profile update but it seems the PrusaSlicer.ini file is not backed up during a snapshot, only the Profiles ?  If so, this seems like a worthwhile change to make to PS in order to aid in future troubleshooting (like this).  Can you confirm, or am I missing something on how to restore prefs fully from a snapshot ?

rtyr, 2020-06-03:
Anyway, I believe we can close this. 3.9.0 is covered :-).

closed 2020-06-03 by rtyr