prusa3d/PrusaSlicer-settings #95: Turn off heatbed before unloading filament (MMU2)

vmeurisse, 2020-06-13:
The current end G-Code for Mk2 MMU2 printers contain this:
```
M702 C

G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
```
The issue is that in case of filament unloading failure, this will leave the the bed heating on, possibly for several hours until someone realize it.

Since the print is effectively finished at this point, there should be no risk turning of heatbed and fan before unloading filament.
```
M140 S0 ; turn off heatbed
M107 ; turn off fan

M702 C

G4 ; wait
M104 S0 ; turn off temperature
```

rtyr, 2020-06-13:
Yes, I agree, we have actually already discussed this change internally.

rtyr, 2020-06-29:
Implemented in 1.1.7. Closing.

closed 2020-06-29 by rtyr