prusa3d/PrusaSlicer-settings #98: Colorfabb XT-CF20 extrusion multiplier

elk0, 2020-06-17:
Hi,
The extrusion multiplier of Colorfabb's XT-CF20 default profile is set to 1.2.
It is way too high for my MK3s.
I currently use it at 1.03.
Maybe it had been set for the single drive extruder of the Mk2?

rtyr, 2020-06-17:
Thank you for your feedback. I am revisiting this profile using current batch of XT-CF20 and hardened 0.4mm nozzle. Current extrusion multiplier seems too high to me as well. I believe that main reason for such multiplier was to prevent clogs on 0.4mm nozzle.

What nozzle are you using? Do you print with default max volumetric speed (very slow)?

elk0, 2020-06-17:
I use a 0.4mm E3d nozzleX.
Yeah, I forgot about that super slow volumetric speed.
I currently use 2.5mm3/s.
I didn't know why it was so low so I didn't want to change it too much in case there was a good reason.

rtyr, 2020-06-29:
Profile is now updated, let's hope for good feedback :-). Closing.

closed 2020-06-29 by rtyr