prusa3d/PrusaSlicer-settings #102: M907 not inserted from Start G-Code Block

BlueHoundInc, 2020-08-16:
### Version
2.2.0

### Operating system type + version
Windows 10 1909 Build 18363.1016

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S 3.9.0 (1.0.6)

### Behavior
With the default printer profile the following is evaluated and M907 E430 is injected to the GCODE as expected.

`; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/(0.05mm ULTRADETAIL @MK3|0.07mm ULTRADETAIL @MK3|0.10mm DETAIL @MK3|0.15mm QUALITY @MK3|0.20mm QUALITY @MK3)/}M907 E430 ; set extruder motor current{endif}`

The print_settings_id at the bottom of the GCODE file is is: print_settings_id=0.20mm QUALITY @MK3

When a project is saved the print settings ID has the filename appended to the front of the print settings id. The value in this case is: print_settings_id=6x Black Lid - Custom Supports.3mf (0.20mm QUALITY @MK3).

The M907 code is not inserted at the beginning of the file.

Interestingly enough, the reset M907 from the END GCODE block is inserted at the end of the code even though the code at the top isn't.

`{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538 ; reset extruder motor current{endif}`

#### Project File (.3MF) where problem occurs

rtyr, 2020-08-16:
That's because end g-code condition is less strict. You are right with the 3MFs though, we will have to modify it. 
Thanks for reporting.

BlueHoundInc, 2020-08-16:
Glad it wasn't just me! I realized the regex in the end was less strict, just wanted to point out it was working as intended. Poor wording on my part.

rtyr, 2020-09-22:
Fixed. Closing.

closed 2020-09-22 by rtyr