prusa3d/PrusaSlicer-settings #104: Add 3DKordo Everfil Filament Profile

feutl, 2020-09-22:
This is a feature request to add the 3DKordo Everfil Filament to the list of supported material.
It is quite easy to get in Europe, most properly because the company is operating from Poland ;) 

The filament seems to have a very good quality.

https://3dkordo.pl

Thanks

rtyr, 2020-09-22:
Do you have profiles/settings for it?

feutl, 2020-09-22:
I used the Generic Ones and made some changes on the device while printing.
Anyhow, as soon as I have my temp tower ready for slicing I start tuning the profiles for PLA and PETG, which I do have at home.
This could take some time, especially because I am not a expert in this matter, perhaps someone could jump on ship too and help getting those things ready for production.