prusa3d/PrusaSlicer-settings #106: Y Axis Crashing with Prusa Mini due to Y end location after each print

DRKMSTR, 2020-10-25:
### Version
_2.2.0_

### Operating system type + version
Windows 10 - Latest

### 3D printer brand / version + firmware version (if known)
Prusa Mini, Firmware V4.2.1

### Behavior
Upon the end of a print, the Y axis moves all the way to the front of the carriage. When any other operation is performed which requires zeroing of the Y axis (starting new print, homing axes, calibrating) the printer forces the Y axis forward to prevent crashing on the rear of the printer, however this instead crashes the Y axis into the front.

Please update to end the print a few millimeters from the front of the Y axis instead of sending it all the way forward.

#### Project File (.3MF) where problem occurs
All

rtyr, 2020-10-27:
I wasn't able to reproduce it on any of 4 random MINIs, but I guess I can move parking position a little.

DRKMSTR, 2020-10-27:
> I wasn't able to reproduce it on any of 4 random MINIs, but I guess I can move parking position a little.

@rtyr I changed the parking location 'Custom G-Code - End G-Code' from Y180 to Y178, this provided enough space for it to stop. 

I tested over 2 days and it works fine. I'm not sure if my printer is just a millimeter too small or if I'm running off an old profile somehow.

rtyr, 2020-11-24:
Changed to X178 Y178. Closing.

closed 2020-11-24 by rtyr