prusa3d/PrusaSlicer-settings #107: Start gcode Some Creality printers prob bed before heated

hdtechk, 2020-11-09:
M190 S[first_layer_bed_temperature] ; wait for bed temp
Needs to be moved before G28 so heated bed is ready before G29 runs.
found when using CR 10s PRO profile and PrusaSlicer 2.3.0 alpha3

rtyr, 2020-11-09:
@pmjdebruijn Could you please look at this?

pmjdebruijn, 2020-11-09:
The reason why this is currently done in parallel, is that it increases print start speed by a fairly large margin. And I've never seen issues because of it at least on an Ender 3.

So the question is, are you actually experiencing first layer issues because of this?

pmjdebruijn, 2020-11-09:
#5114 will probably offer a reasonably compromise

hdtechk, 2020-11-09:
Thanks for looking at this. Yes when I manually moved the M190 it made big differences in my first layer.

closed 2020-11-09 by hdtechk