prusa3d/PrusaSlicer-settings #111: Include pre-built configuration for more nozzle sizes (e.g: 0.8 mm)

ffcruz85, 2020-12-19:
### Version
2.2.0 (win64)

### Operating system type + version
Windows 10 (not related)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S - firmware 3.9.1

### Behavior
Built-in printer configuration profiles for Prusa printers only include 0.25, 0.4, and 0.6mm nozzles. It would be great if we could get all possible nozzle sizes available for v6 nozzles, namely 0.15, 0.5, 0.8 mm

Sure, there are some prebuilt profiles shared online (not even that many), but it's hard for a beginner/intermediate user to know they are using the correct settings for these other size nozzles, even more so for the nozzle/printer combination. In any case, it would allow people to more easily experiment with different nozzles without running into all sorts of potential issues - with the confidence that it is properly configured by the Prusa pros. Maybe there's a reason not to have them, but I just don't see why not.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
none

rtyr, 2020-12-19:
0.8mm nozzle profiles will be added soon.
0.5mm - possibly in the future
0.15mm - unlikely

ffcruz85, 2021-02-23:
Sorry, I missed your comment - I am already using it :) If I may, just a follow up question, related to the other issue you closed.  Will the different nozzle sizes also be supported in the firmware? As in being able to select it in the hardware menu and do first layer calibration?
Thanks! You can close the issue, just wanted to ask about if there will be corresponding changes in firmware

rtyr, 2021-02-24:
Yes, the 0.8mm nozzle option will be added to the next firmware version. It currently only checks if the gcode was sliced for correct nozzle diameter. First layer calibration is hardcoded for 0.4mm nozzle at the moment (nozzle diameter setting is ignored), so it is better to slice some simple object for different nozzle sizes.

rtyr, 2021-02-24:
We also discussed a possibility to add user defined custom nozzle sizes. I believe it is on the "todo" list of firmware team.

ffcruz85, 2021-02-24:
Thank you! And 5he slicer profile has already been a huge help!

closed 2021-02-24 by ffcruz85