prusa3d/PrusaSlicer-settings #113: add igus Tribofilament iglidur I150 setting

Igami, 2021-01-04:
> igus is donating a total of 1,000 spools of the wear-resistant iglidur® I150 filament to charitable organizations that are committed to repairing defective everyday objects.

https://content.communication.igus.net/3d-druck-ersatzteile-filament-spende

I've tested the filament with the original Prusa MK3S+MMU2S with the settings below and will still test it with the Anycubic Mega (S) this week.
After that I will make a pull request with the settings.

```ini
[filament:iglidur® I150-PF-0175-0750]
bed_temperature = 60
filament_colour = #EEEEEC
filament_cost = 62.46
filament_density = 1.3
filament_max_volumetric_speed = 2
filament_notes = "Easy to process, abrasion-resistant filament - suitable for food contact\n\niglidur I150-PF is characterised by a high wear resistance as the material is enriched with solid lubricants. It can be \nprocessed just as easily as standard PLA and PETG filaments. Due to its high wear resistance and good mechanical \nproperties (strength, toughness, layer adhesion), it is an all-rounder material that is suitable for most applications and 3D \nprinters. Furthermore, it is suitable for food contact due to its conformity to EU 10/2011 and has proven itself accordingly \nin applications in the food and packaging industry. As a versatile all-rounder, iglidur I150 is by far the most frequently \nused igus filament. There are only restrictions in the maximum permanent application temperature of 65°C and for highly \ndynamic applications. With the igus Bonding agent for tribo-filaments, the iglidur I150 can be processed on conventional \nprint bed surfaces (e.g. glass).\n\nhttps://www.igus.eu/product/12404"
filament_spool_weight = 230
filament_type = TRIBO
filament_vendor = igus
first_layer_bed_temperature = 60
first_layer_temperature = 250
temperature = 240
```

Markus-Be, 2022-02-24:
Did you tested this settings ?

Igami, 2022-02-27:
I retested it. It works on the smooth sheet but there is a lot of stringing.
[image]

egnor, 2023-12-19:
(I know this has been sitting there for a couple years, but I found it...)

- you set the filament type to `TRIBO`, which isn't one of the standard values-- when I go to print with that, the printer always complains that the wrong type of filament is loaded?
- you set the max volumetric speed to 2 g/mm3 but the data sheet gives the number 25?
- you set first layer nozzle temp to 250C but the data sheet gives a max temp of 240C?
- any empirical or theoretical reason for the specific temperatures?