prusa3d/PrusaSlicer-settings #114: "Extra Loading Distance" Setting for MK3s + MMU2s

carterjb, 2021-01-18:
I am having an issue with the "Extra Loading Distance" feature within PrusaSlicer for my MK3s + MMU2s printer (Printer Settings -> Single extruder MM setup -> Extra Loading Distance). I have the Mosquito hotend and thermistor, that is the only difference for my printer. I think my customizations may have possibly changed the distance between the extruder gears and nozzle slightly. I do not believe this is the root cause of my issue with this setting, however, please read on. 

I believe there is an issue when loading after the filament reaches the IR sensor, it is driven slightly too far into the hotend leading to a blob every filament change. Please note, that I have tuned my filament changes to work quite perfectly other than this (Cooling moves, sensor calibration, etc.). These blobs are large enough to cause nozzle collisions and skipped steps within the wipe tower. Reducing "Filament Parking Position" seems to solve this issue, but causes another issue because upon unloading the filament isn't driven far enough.

There appears to be a feature in place for this exact problem, called "Extra Loading Distance", which can be negative. This would be a perfect setting for my problem, however, making the value negative doesn't seem to shorten the initial distance the filament is driven. Instead, I am observing it drive the filament forward by "Filament Parking Distance" first, then adjust by "Extra Loading Distance". However, this sequential retraction for negative values of "Extra Loading Distance" is not resolving my issue. Essentially, I would like it to adjust the true initial loading distance, such that the remaining filament in the hotend isn't accidently extruded creating this blobbing issue. 

For example, I have even tried "Filament Parking Distance" = 85mm and "Extra Loading Distance" = -75mm as an extreme case. I observed it load the filament 85mm creating a blob, then retract 75mm such that the hotend didn't extrude anything for a long time during the wipe and print (obviously there was a void in the hotend). This is definitely not how I would need this setting to work in order to resolve my problem. In this extreme case, I would like it to instead just drive filament by 10mm toward the nozzle, which shouldn't create a blob. Then, I could tweak this setting such that remaining filament in the hotend won't cause these blobs while loading.

rtyr, 2021-01-23:
As far as I know, you need modified firmware (and printer profile). I don't now exact details, values, etc. I also don't know how reliably this hotend works with MMU2. You should contact Slice Engineering about particular steps needed to succesfully use their hotend with MMU2.

This thread may help you https://www.facebook.com/groups/prusacommunity/permalink/1176229959384401.

Closing.

closed 2021-01-23 by rtyr

carterjb, 2021-01-23:
I believe there is possibly an issue with this setting ("Extra Loading Distance"). It seems that it does not truly reduce the amount that the filament is loaded into the hotend when set to negative values. I believe this alone would solve my problem. I simply need the "Extra Loading Distance" to reduce the amount the filament is loaded into the hotend initially (while the hotend is stationary). This would solve my blobbing issue. I believe from the description it is supposed to do that, but it does not seem to be working properly. I was hoping someone could explain why this is the case, or how I could fix it?

rtyr, 2021-01-23:
No, the position is also hardcoded in the firmware. Firmware expects stock hardware.

carterjb, 2021-01-25:
My main point of confusion is that within the MK3s+MMU2s profile in PrusaSlicer, I can change the "Filament Parking Position" and "Extra Loading Distance" parameters. I have verified that these settings change the behavior of the printer, so I am unsure how this interacts with the firmware settings (maybe priority is given to PrusaSlicer)? 

My second point of confusion is I would like to know where the "Extra Loading Distance" functionality is coded? If this is coded in the Firmware, then I believe changing that should suffice. However, if PrusaSlicer is controlling this behavior, then I believe my issue cannot be solved regardless of the "Filament Parking Distance". Essentially, I will still need to somehow resolve the issue with "Extra Loading Distance" acting as a sequential action, rather than a modifier setting to the initial loading distance. This will allow me to reduce the loading distance to adapt to residual filament left in the hotend. My thought is the Mosquito may just be more prone to having residual filament left in the hotend.