prusa3d/PrusaSlicer-settings #115: Filament system presets disagree with datasheets

Patricol, 2021-01-19:
Maybe it's bad luck; but many of the profiles I've looked at disagree with datasheets on density etc.

For example, Hatchbox PLA's density is 1.27 according to their [data sheet](https://cdn.shopify.com/s/files/1/0008/2457/4018/files/HATCHBOX_PLA_3D_Printer_Filament_SDS.pdf?1561), but 1.24 in the system preset.

Verbatim BVOH's density is further off; [1.14](https://www.verbatim-europe.co.uk/en/prod/verbatim-bvoh-filament-175-mm--transparent-55903/) officially and 1.23 in the system preset.
Verbatim BVOH's price is also way off; 79.99/1kg in the system preset, and [79.99/500g
](https://shop.prusa3d.com/en/special/161-verbatim-bvoh-soluble-support-05kg.html) from Prusa's store.

Are these expected to be accurate? Do the datasheets just change too frequently?

Thanks!

rtyr, 2021-01-19:
Hi, I will look into it. Thanks.

rtyr, 2021-02-19:
Fixed in configuration bundle 1.2.4. Density values are often hard to find or not provided by manufacturer, so there may still be some inaccurate values. Prices are USD+vat. Closing.

closed 2021-02-19 by rtyr