prusa3d/PrusaSlicer-settings #116: Many filaments listed as not being compatible with the MMU2S

nirvdrum, 2021-01-19:
Many of the filaments that are compatible with the MK3S are listed as being incompatible with the MK3S+MMU2S for multi-extrude mode. Given it's virtually the same printer, it looks like this might just be a bug. The net effect though is the default filament options available are limited to Prusament and Generic variants of the filament types.

rtyr, 2021-01-19:
Most of the filament profiles are available for MMU2 in single mode, but only some selected filaments are available for multimaterial mode. It is intended. These profiles have similar name, but they are different from standard profiles (lower temps, specific MMU2 ramming settings, etc.).

Closing.

closed 2021-01-19 by rtyr

nirvdrum, 2021-01-19:
Thanks for clarifying. Based on discussions with other MMU2S owners, it's pretty common to run single color/material prints is multi-extruder mode so you don't have to select the filament from the printer (useful for PrusaSlicer sending to Octoprint). So, I think there's still a valid use case here. But, if there's no chance of this getting changed, I'll just modify my configuration locally.