prusa3d/PrusaSlicer-settings #119: Ender 3 V2 End G-code issue

nanortemis, 2021-02-11:
Hi, 

After updating the Creality to 0.0.13 I'm getting this at the end of a produced gcode file:

```gcode
;END gcode for filament
G1 Z5.08 F600 ; Move print head up
G1 X5 Y176 F9000 ; present print
G1 Z73.08 F600 ; Move print head further up
G1 Z150 F600 ; Move print head further up
M140 S0 ; turn off heatbed
M104 S0 ; turn off temperature
M107 ; turn off fan
M84 X Y E ; disable motors
```

Is this expected? the print head goes to 73.08 then to 150 and at the end motors are disabled and my print head is dropping down.

Object details:
[image]

rtyr, 2021-02-11:
@pmjdebruijn Can you please look into it?

pmjdebruijn, 2021-02-11:
Since 0.0.13 there is a heuristic in the end_gcode that will raise Z to at least 60% of the printers height, this is to make low prints easier to remove, as the standard +70mm move can sometimes still end up interfering with print removal. But we don't want to just replace the +70mm with a higher value, as it's pointless for larger prints (and make starting the next print take longer), aside from unnecessarily straining the bowden tube and cabling. It's implemented as two seperate moves to keep the end_gcode much more easy to understand.

As far as turning the motors off is concerned, nothing has changed, I've never touched the M84 line as far as I can recall. Also do notice that the Z axis is specifically NOT turned off in the generated gcode.

So nothing changed that could be the root cause for your dropping print head issue. I'm guessing you probably have your eccentric nuts too loose, where at some point gravity overpowers the steppers when they eventually get powered down. Some firmwares have a timeout on keeping the steppers powered if I'm not mistaken. You may not have noticed this earlier because of the height difference, as the higher the gantry is, it also starts bearing the weight of the cabling more significantly, which may contribute to crossing that threshold.

rtyr, 2021-04-22:
I guess we can close this.

closed 2021-04-22 by rtyr