prusa3d/PrusaSlicer-settings #120: 0 filament density with 0,6 nozzle on MINI

Pitel, 2021-02-11:
### Version
2.3.0

### Operating system type + version
Ubuntu, but shouldn't matter

### 3D printer brand / version + firmware version (if known)
MINI

### Behavior
* Start from clean install, in wizard select mini with 0,6 mm nozzle and make sure Generic PETG and Plasty Mladec PETG filaments are chosen. Finish wizard.
* Add cube, or anyu STL model, doesn't matter.
* Select 0,6 MINI printer and Generic PETG filament.
* The weight of filament in print stats is missing, because with this configuration, the filament density is set to 0.

Probably an error in default print profiles?

* This happens with Generic and Plasty Mladec PETG. Prusament PETG is fine, PLAs are fine.
* 0,4 nozzle and 0,8 nozzle are ok.
* Different printers are also ok.

rtyr, 2021-02-11:
You are right. Thanks, I will fix it.

rtyr, 2021-02-19:
Fixed in 1.2.4. Closing.

closed 2021-02-19 by rtyr