prusa3d/PrusaSlicer-settings #123: missing 0.25 nozzle MMU profiles

cabbey, 2021-02-16:
The current ini file is missing the MMU2 and MMU2S profiles for the 0.25nozzle. I've worked up profiles for them and done quite a number of prints on my Mk2.5S + MMU2S and they perform no worse than the 0.4 or 0.6 nozzle profiles... in fact the may perform slightly better as the smaller nozzle doesn't ooze quite as much when the old filament is removed. I've worked out what I think is a patch to the ini file to generate the same profile as I've been testing with, but I don't see a good way to test it to confirm. I would submit a PR but the current way of handling the ini file in git doesn't seem to be conducive to that. Also note that I don't have access to a Mk3, so I can't test my profiles there; thus I've not patched in support for the Mk3 family. But conceptually, the changes should be similar... source the 0.4 nozzle MMU2 profile, change the 4 things that are dependent on the nozzle diameter and the default print profile to pick... all of which can be gleaned from the corresponding "MMU Single" profile.

[prusa-settings.patch.gz](https://github.com/prusa3d/PrusaSlicer-settings/files/5986862/prusa-settings.patch.gz)

rtyr, 2021-02-16:
It is intended. Current ramming sequences are not suitable for 0.25mm nozzle (high volumetric speed).

cabbey, 2021-02-27:
@rtyr hmm... I've done quite a number of prints successfully with this setup. Is there something in particular you've seen with the tip re-shaping that's not quite right? Or a particular setting that is believed to need some tuning? I'm happy to experiment and see if there's improvements to be had, but at the moment I'm not seeing any obvious issues. That said, I'm also generally using a filament profile that's not public, perhaps with more common filament profiles there are issues I'm not seeing? The only prusament I have access to is either PETG or has stuff in it (so not a good idea to use with the 0.25 nozzles)... but I have a few spools of good old Plasty Mladec PLA around... would I see the issues with that filament and the standard profile for it?

rtyr, 2022-04-27:
MMU profiles for 0.25mm nozzle were just added via 1.4.5 configuration update. Closing.

closed 2022-04-27 by rtyr