prusa3d/PrusaSlicer-settings #124: Ghosting problems when using PrusaSlicer–generated G-Code, on Prusa MINI

PaperVolume, 2021-02-16:
### Version
_2.3.0_

### Operating system type + version
_Windows 10, 64-bit & 32-bit #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Original Prusa MINI, partially assembled, with a 0.25mm nozzle, on firmware 4.2.1 & 4.3.0 #s_

### Behavior
* _Severe ghosting appears when model is printed and sliced with PrusaSlicer. Cura 3D has no such problem. As far as I know this is completely independent from seam position_
* _Every G-Code generated with PrusaSlicer exhibits this behavior so far._
* _Ghosting always visible on supports, with intensity increasing as layer height decreases (none on 0.15mm, visible on 0.07mm)_
* _As expected_
  * _[image]_

#### Project File (.3MF) where problem occurs (example)
_[100_StrikeDagger_LowerArm_SlideArm_01.zip](https://github.com/prusa3d/PrusaSlicer/files/5987812/100_StrikeDagger_LowerArm_SlideArm_01.zip)_
_[image]
_

PaperVolume, 2021-02-18:
@rtyr Thank you for your reply. I have increased extrusion width from 0.2mm to 0.3mm, but increasing extrusion multiplier seemed to yield better results on supports.

rtyr, 2021-02-18:
Maybe your MINI is underextruding a little bit.

PaperVolume, 2021-02-18:
Right after this problem first occured I made every hardware fix I could find. I found my way here through Prusa support and a common assumption was that this is indeed a slicer issue, considering Cura doesn't show these problems. Previously these settings weren't doing anything out of normal, which is weird because this indicates a maintenance issue, and they were all redundantly adressed.

rtyr, 2021-02-18:
It is not so easily comparable, since PrusaSlicer uses variable layer height for the supports, independent from main object's layer height. If it printed fine before, then likely something has changed on the printer side.

closed 2021-02-18 by PaperVolume