prusa3d/PrusaSlicer-settings #132: Add „das filament PETG“ to PrusaResearch

northerx, 2021-05-01:
The Filament „das filament PETG“ is available for creality printers. So print property's should be known ;) Unfortunately not yet for Prusa printers.

rtyr, 2021-05-01:
> So print property's should be known ;)

That profile was added by community, specifically for Creality printers. Closing.

rtyr, 2021-05-01:
That being said, I will likely add profile for this filament in the future.

closed 2021-05-01 by rtyr

rtyr, 2021-09-02:
This filament was not yet added for our printers.

rtyr, 2021-09-02:
Tested filament profile for our printers would certainly help. Recently added filament profiles for add:north and Extrudr filaments were provided directly by manufacturer.

Skaronator, 2025-07-09:
Just usw Generic PETG. Custom profiles are mostly overrrated IMO and Generic works 99% just as well.