prusa3d/PrusaSlicer-settings #139: Prusa Mini default End G-code max height error

nsc-jens, 2021-08-12:
### Version
PrusaSlicer 2.3.1
Prusa Mini profile 1.2.8

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Bug report
When a print ends the hotend is lifted 2 mm unless the printer max z is reached:

`{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)}{endif} F720 ; Move print head up`

There is a logic error here if printing a model that is exactly 180 mm high. If max_layer_z  = max_print_height = 180, then the content of the if statement never gets to the resulting gcode (correct) and only the " F720 ; Move print head up" part makes it there (wrong). The "{endif}" could be moved to the end of the line. Like: 

`{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F720 ; Move print head up{endif}`

There is another similar if statement a few lines lower down.

The test is ultimately not needed I think. It could probably be simplified to:

`G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F720 ; Move print head up`

This would limit the upwards movement to max_print_height and be a non-movement if already at 180 mm. I have not tested this and I don't know if there are corner cases where max_layer_z actually can be bigger than max_print_height.

rtyr, 2021-08-16:
It doesn't cause any real issue, but I will change it in the next update. Thanks.

rtyr, 2021-09-13:
Fixed in configuration bundle 1.3.2. Closing.

closed 2021-09-13 by rtyr